mavlink/qgroundcontrol #12138: Cannot be compiled

mehmetdrnmrkt, 2024-11-26:
Hello, I installed the requirements by following the build instructions [here.](https://docs.qgroundcontrol.com/master/en/qgc-dev-guide/getting_started/)
But when I try to open it in debug mode I get an error. It can't be compiled correctly. Maybe it's something very simple but I couldn't find a solution. Here are some errors in cmake output

`
.
.
.

[cmake] -- Could NOT find WrapVulkanHeaders (missing: Vulkan_INCLUDE_DIR) 
[cmake] -- Could NOT find WrapVulkanHeaders (missing: Vulkan_INCLUDE_DIR) 
[cmake] -- Could NOT find Qt6WaylandClient (missing: Qt6WaylandClient_DIR)
[cmake] CMake Warning at C:/Qt/6.6.3/msvc2019_64/lib/cmake/Qt6/Qt6Config.cmake:221 (message):
[cmake]   Failed to find optional Qt component "WaylandClient".

.
.
.

[cmake]   "C:/Qt/6.6.3/msvc2019_64/lib/cmake/Qt6WaylandClient/Qt6WaylandClientConfig.cmake"
[cmake]   does NOT exist
[cmake] 
[cmake] Call Stack (most recent call first):
[cmake]   CMakeLists.txt:157 (find_package)
[cmake] 
[cmake] 
[cmake] -- Could NOT find LibExiv2: Found unsuitable version "..", but required is at least "0.28.2" (found LibExiv2_LIBRARIES-NOTFOUND)
[cmake] -- Could NOT find exiv2 (missing: exiv2_DIR)
[cmake] -- Building Exiv2
[cmake] -- Could NOT find Iconv (missing: Iconv_LIBRARY Iconv_INCLUDE_DIR) 

`
The whole output is like this. [cmake_output.txt](https://github.com/user-attachments/files/17919324/cmake_output.txt)

HTRamsey, 2024-11-26:
The errors you listed are actually okay, the problem is later on with gstreamer. Do you have you have gstreamer runtime and developer of at least version 1.22 installed?

HTRamsey, 2024-11-27:
Try switching the GIT_TAG from v0.28.3 to main in the following code from AnalyzeView->CMakeLists.txt
```
FetchContent_Declare(EXIV2
        GIT_REPOSITORY https://github.com/Exiv2/exiv2.git
        GIT_TAG v0.28.3
        GIT_SHALLOW TRUE
    )
```

HTRamsey, 2024-11-27:
I was aware of the issue with Exiv2 & got a fix committed to that git repo, but I'm not sure why it was a critical error for you instead of just a warning

closed 2024-11-27 by mehmetdrnmrkt