mavlink/qgroundcontrol #12144: Docs: page is missing

hbaklan943, 2024-11-27:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/support/support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
QGC docs says that we can chose the used dialect my modifying MAVLINK_CONF. it should redirect to proper location where MAVLINK_CONF defined.

## Current Behavior
MAVLINK_CONFIG redirects to a non existing page.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. go to https://docs.qgroundcontrol.com/master/en/qgc-dev-guide/custom_build/mavlink.html
2. on the bottom (You can change the whole dialect used by setting it in [MAVLINK_CONF](https://github.com/mavlink/qgroundcontrol/blob/master/QGCExternalLibs.pri#L52) when running qmake.) click the link 

## System Information
When posting bug reports, include the following information
- Operating System: [Ubuntu 22.04]
- QGC Version: []
- QGC build: [e.g., daily, stable, self-built from source, etc...]
- Flight Controller: [Pixhawk 2,]
- Autopilot (with version): [PX4]

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
[image]

hamishwillee, 2024-11-27:
@DonLakeFlyer This is correct - `MAVLINK_CONF` used to be defined in QGCExternalLibs.pri, but that file no longer seems to exist.

HTRamsey, 2024-11-27:
Needs to be updated for CMake, it will instead be CustomOptions.cmake

hbaklan943, 2024-11-27:
Also the same [section of the page](https://docs.px4.io/main/en/middleware/mavlink.html#updating-qgroundcontrol) says that 
"QGC uses a pre-built C library that must be located at [/qgroundcontrol/libs/mavlink/include/mavlink](https://github.com/mavlink/qgroundcontrol/tree/master/libs/mavlink/include/mavlink) in the QGC source."
but even after I delete those pre-built C libraries and launch the qgc, it still continues to successfully receive mavlink messages.
There seems to be another problem about this too. 

Another question if anyone willing to help: Where are the mavlink message definitions located at in qgc source code, I couldn't find it.

hbaklan943, 2024-11-27:
new findings from me. tldr; i guess when building qgc it fetches pre-built mavlink libraries from [a git repo](https://github.com/mavlink/c_library_v2)

[cmake/CustomOptions.cmake](https://github.com/mavlink/qgroundcontrol/blob/fe917b39fe0cd2be096599b9e066a41960994a57/cmake/CustomOptions.cmake)
```cmake
set(QGC_MAVLINK_GIT_REPO "https://github.com/mavlink/c_library_v2.git" CACHE STRING "URL to MAVLink Git Repo")
set(QGC_MAVLINK_GIT_TAG "4db2f67156d996eae90ef437a73353468d850407" CACHE STRING "Tag of MAVLink Git Repo")
```

[src/MAVLink/CMakeLists.txt](https://github.com/mavlink/qgroundcontrol/blob/fe917b39fe0cd2be096599b9e066a41960994a57/src/MAVLink/CMakeLists.txt)
```cmake
FetchContent_Declare(mavlink
    GIT_REPOSITORY ${QGC_MAVLINK_GIT_REPO}
    GIT_TAG ${QGC_MAVLINK_GIT_TAG}
)
FetchContent_MakeAvailable(mavlink)

target_include_directories(MAVLink
    PUBLIC
        ${mavlink_SOURCE_DIR}
        ${mavlink_SOURCE_DIR}/all
        ${mavlink_SOURCE_DIR}/common
)
```

I will try to make it fetch from a repo that i created with my custom message definitions. I it works i may try to update docs.

hamishwillee, 2024-11-28:
Thanks @hbaklan943 Even if it fetches the repo, you'd still need to select the dialect to use. So that must be specified somewhere.

hbaklan943, 2024-11-28:
> Thanks @hbaklan943 Even if it fetches the repo, you'd still need to select the dialect to use. So that must be specified somewhere.

I couldn't find where the selected dialect is specified but i didn't had to set anything, I just changed the fetched git repo link and the tag of that repo. 

It seem's this is automated in somewhere but i couldn't find where that happens. For now maintaining a git repo for my custom mesages solves my problem.

Also I guess even though we find where the c-headers and definitions located(probably under `libs/mavlink/include/mavlink`) and change them they will be replaced with the fetched repo on each build.

HTRamsey, 2024-11-28:
@hbaklan943 I have not finished with all the custom options like selecting mavlink dialect in cmake yet. But I've been working on it and will add that option soon