mavlink/qgroundcontrol #12162: qgroundcontrol package

50ke, 2024-12-01:
# Env
- QT: 6.6.3
- branch: master
- OS: Windows 10
- compiler: msvc2019 64bit
# Question
I can compile and run the QGC source code in release mode normally. (1) I used windeployqt6 for packaging, but when I run QGroundControl.exe, there is no response and no error messages. (2) I tried using Inno Setup, but after packaging and installation, running it reports a missing dll. Could you please provide some guidance on how to package QGC into an executable exe file? Thank you very much.

[image]
[image]

=============QGC.iss Content Start==============
; Script generated by the Inno Setup Script Wizard.
; SEE THE DOCUMENTATION FOR DETAILS ON CREATING INNO SETUP SCRIPT FILES!

#define MyAppName "QGroundControl"
#define MyAppVersion "1.0"
#define MyAppPublisher "My Company, Inc."
#define MyAppURL "https://www.example.com/"
#define MyAppExeName "QGroundControl.exe"
#define MyAppAssocName MyAppName + ""
#define MyAppAssocExt ".myp"
#define MyAppAssocKey StringChange(MyAppAssocName, " ", "") + MyAppAssocExt

[Setup]
; NOTE: The value of AppId uniquely identifies this application. Do not use the same AppId value in installers for other applications.
; (To generate a new GUID, click Tools | Generate GUID inside the IDE.)
AppId={{CBC36EFE-797C-48AA-B29A-E9751181ECA9}
AppName={#MyAppName}
AppVersion={#MyAppVersion}
;AppVerName={#MyAppName} {#MyAppVersion}
AppPublisher={#MyAppPublisher}
AppPublisherURL={#MyAppURL}
AppSupportURL={#MyAppURL}
AppUpdatesURL={#MyAppURL}
DefaultDirName=D:\latest\package\addition\{#MyAppName}
; "ArchitecturesAllowed=x64compatible" specifies that Setup cannot run
; on anything but x64 and Windows 11 on Arm.
ArchitecturesAllowed=x64compatible
; "ArchitecturesInstallIn64BitMode=x64compatible" requests that the
; install be done in "64-bit mode" on x64 or Windows 11 on Arm,
; meaning it should use the native 64-bit Program Files directory and
; the 64-bit view of the registry.
ArchitecturesInstallIn64BitMode=x64compatible
ChangesAssociations=yes
DisableProgramGroupPage=yes
LicenseFile=D:\latest\package\addition\license.txt
InfoBeforeFile=D:\latest\package\addition\before.txt
InfoAfterFile=D:\latest\package\addition\after.txt
; Uncomment the following line to run in non administrative install mode (install for current user only.)
;PrivilegesRequired=lowest
OutputDir=D:\latest\package\output
OutputBaseFilename=QGroundControlInstaller
SetupIconFile=D:\latest\package\addition\ugc.ico
Compression=lzma
SolidCompression=yes
WizardStyle=modern

[Languages]
Name: "english"; MessagesFile: "compiler:Default.isl"

[Tasks]
Name: "desktopicon"; Description: "{cm:CreateDesktopIcon}"; GroupDescription: "{cm:AdditionalIcons}"; Flags: unchecked

[Files]
Source: "D:\latest\dev\qgroundcontrol\build\Desktop_Qt_6_6_3_MSVC2019_64bit-Release\{#MyAppExeName}"; DestDir: "{app}"; Flags: ignoreversion
Source: "D:\latest\dev\qgroundcontrol\build\Desktop_Qt_6_6_3_MSVC2019_64bit-Release\*"; DestDir: "{app}"; Flags: ignoreversion recursesubdirs createallsubdirs
; NOTE: Don't use "Flags: ignoreversion" on any shared system files

[Registry]
Root: HKA; Subkey: "Software\Classes\{#MyAppAssocExt}\OpenWithProgids"; ValueType: string; ValueName: "{#MyAppAssocKey}"; ValueData: ""; Flags: uninsdeletevalue
Root: HKA; Subkey: "Software\Classes\{#MyAppAssocKey}"; ValueType: string; ValueName: ""; ValueData: "{#MyAppAssocName}"; Flags: uninsdeletekey
Root: HKA; Subkey: "Software\Classes\{#MyAppAssocKey}\DefaultIcon"; ValueType: string; ValueName: ""; ValueData: "{app}\{#MyAppExeName},0"
Root: HKA; Subkey: "Software\Classes\{#MyAppAssocKey}\shell\open\command"; ValueType: string; ValueName: ""; ValueData: """{app}\{#MyAppExeName}"" ""%1"""
Root: HKA; Subkey: "Software\Classes\Applications\{#MyAppExeName}\SupportedTypes"; ValueType: string; ValueName: ".myp"; ValueData: ""

[Icons]
Name: "{autoprograms}\{#MyAppName}"; Filename: "{app}\{#MyAppExeName}"
Name: "{autodesktop}\{#MyAppName}"; Filename: "{app}\{#MyAppExeName}"; Tasks: desktopicon

[Run]
Filename: "{app}\{#MyAppExeName}"; Description: "{cm:LaunchProgram,{#StringChange(MyAppName, '&', '&&')}}"; Flags: nowait postinstall skipifsilent
=============QGC.iss Content End==============

HTRamsey, 2024-12-14:
This is how we do it currently: https://github.com/mavlink/qgroundcontrol/blob/master/cmake/CreateWinInstaller.cmake
But I'd like to switch it to use the Qt Installer Framework soon

hamishwillee, 2024-12-18:
Whatever gets done, can we add docs at the same time as the work is done.

HTRamsey, 2024-12-18:
Yeah I'll talk with Don on what needs updating. Wanted to avoid having to rewrite docs on things likely to change but it would definitely cut down on posted issues

closed 2025-04-22 by HTRamsey