mavlink/qgroundcontrol #12171: Cannot be compiled

mpnemo, 2024-12-05:
I got stuck here

[cmake] CMake Error in src/Joystick/CMakeLists.txt:
**[cmake]   The dependency target "PkgConfig::ZLIB" of target
[cmake]   "Joystick_autogen_timestamp_deps" does not exist.**
[cmake] 
[cmake] 
[cmake] -- Generating done (3.6s)
[cmake] CMake Generate step failed.  Build files cannot be regenerated correctly.

It does find ZLIB
[cmake] -- Not enabling format target (clang-format not found)
[cmake] -- Could NOT find ZLIB (missing: ZLIB_LIBRARY ZLIB_INCLUDE_DIR) (Required is at least version "1.3")
[cmake] -- Checking for module 'ZLIB>=1.3'
[cmake] --   Found ZLIB, version 1.3.1
[cmake] -- Found ZLIB 1.3.1

any ideas?

HTRamsey, 2024-12-05:
Can you  try the changes on the lines marked yellow? in Utilities/Compression/CMakeLists.txt?
[image]

mpnemo, 2024-12-05:
this didnt work. I scrolled up a bit and copied the entire error log below, not sure if that is related to the autogen error above?
[cmake] CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
[cmake] Please set them or make sure they are set and tested correctly in the CMake files:
[cmake] E:/source/qgroundcontrol/build/Desktop_Qt_6_6_3_MSVC2019_64bit-Debug/_deps/exiv2-src/src/ZLIB_INCLUDE_DIR
[cmake]    used as include directory in directory E:/source/qgroundcontrol/build/Desktop_Qt_6_6_3_MSVC2019_64bit-Debug/_deps/exiv2-src/src
[cmake] 
[cmake] CMake Warning (dev) in build/Desktop_Qt_6_6_3_MSVC2019_64bit-Debug/_deps/geographiclib-src/src/CMakeLists.txt:
[cmake]   Policy CMP0100 is not set: Let AUTOMOC and AUTOUIC process .hh files.  Run
[cmake]   "cmake --help-policy CMP0100" for policy details.  Use the cmake_policy
[cmake]   command to set the policy and suppress this warning.
[cmake] 
[cmake]   For compatibility, CMake is excluding the header file(s):
[cmake] 
[cmake]     "E:/source/qgroundcontrol/build/Desktop_Qt_6_6_3_MSVC2019_64bit-Debug/_deps/geographiclib-src/src/kissfft.hh"
[cmake] 
[cmake]   from processing by AUTOMOC and AUTOUIC.  If any of the files should be
[cmake]   processed, set CMP0100 to NEW.  If any of the files should not be
[cmake]   processed, explicitly exclude them by setting the source file property
[cmake]   SKIP_AUTOGEN:
[cmake] 
[cmake]     set_property(SOURCE file.hh PROPERTY SKIP_AUTOGEN ON)
[cmake] 
[cmake] This warning is for project developers.  Use -Wno-dev to suppress it.
[cmake] 
[cmake] CMake Error in src/Joystick/CMakeLists.txt:
[cmake]   The dependency target "PkgConfig::ZLIB" of target
[cmake]   "Joystick_autogen_timestamp_deps" does not exist.

HTRamsey, 2024-12-05:
oh it's the exiv2 thing. Go to where it says 
```
FetchContent_Declare(EXIV2
        GIT_REPOSITORY https://github.com/Exiv2/exiv2.git
        GIT_TAG v0.28.3
        GIT_SHALLOW TRUE
    )
```
and switch GIT_TAG to main

HTRamsey, 2024-12-05:
I submitted a bug fix to them but it's not yet in a release

mpnemo, 2024-12-06:
I tried that, still throws the same cmake error . searched the build for the exiv2.git include, see screenshot at the bottom...

not sure what causes this error, i removed the entire repo, cloned it and did only the changes from your commit here https://github.com/mavlink/qgroundcontrol/pull/12175/commits/4cb5dc5d37124f0036e4fa4e6c3ceb529fe2e0a4

[image]

mpnemo, 2024-12-07:
forget about it. got it to work. i had zlib installed via vcpkg in the environment, this screwed up the environment. pity there is no docker type stuff for windows.

closed 2024-12-07 by HTRamsey