mavlink/qgroundcontrol #12174: Android: Vehicle Setup page crashes app

k-kleber, 2024-12-05:
## Current Behavior
When accessing the Vehicle Setup page the app crashes.
When disconnected, it always crashes.
When connected to a vehicle it works after the second attempt, crashes once, reopen app then works.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Just have the android app open without being connected to a vehicle and select the Vehicle Setup Page.

## System Information
When posting bug reports, include the following information
- Operating System: Android 9
- QGC Version: 99a7c6fcb73f9848a1a943afcef3280fec3ad806 (master)
- QGC build: daily
- Flight Controller: Independent
- Autopilot (with version): Ardupilot 4.5.3

HTRamsey, 2024-12-05:
Hmm not able to reproduce this

DonLakeFlyer, 2024-12-05:
I just go the Herelink version updated to daily and I'm not seeing it there either

k-kleber, 2024-12-09:
i managed a temp fix of this by not having the summary page load initially, just when clicking the summary button. As this only happens on the very old Android 9 system I don't think it makes sense to create a PR for this fix, but leaving the solution here in case someone else struggles with it.

HTRamsey, 2024-12-09:
Probably should use more loaders for things like that in general but it makes the QML a lot more messy and harder to read

DonLakeFlyer, 2024-12-09:
@k-kleber So this is happening when QGC does the automatic switch to vehicle setup when there is setup still not done?

@HTRamsey That's also the case which causes the dialog parenting problem isn't it?

Might be simplest to just remove that automatic view switch and fix both of these. Sometimes that view switfch thing is helpful other things it's just flat out annoying anyway.

HTRamsey, 2024-12-09:
It's one of them. But there are several more ways to trigger the dialog issue. It seems to me this issue was any time accessing the vehicle setup page rather than the automatic switch

k-kleber, 2024-12-10:
Hey, this happens when the vehicle setup is selected, not through an automatic switch. 

The summary is then selected per Default which makes it crash the first time after a reboot.

closed 2026-10-04 by github-actions[bot]