mavlink/qgroundcontrol #12182: Build issue with the latest veriosn

omid-esrafilian, 2024-12-09:
I have just rebased to the latest version, and since then, I am unable to build the code on either Linux or Windows.

In Linux, I get the following error:
error: Imported target "GStreamer::GStreamer"

And in Windows, this error:
error: Could NOT find GLIB2

It seems that the issue is related to GStreamer, which used to be optional. 
I'm wondering if there is a way to exclude the GStreamer requirement when compiling the 

Thanks

HTRamsey, 2024-12-09:
It's still optional through QGC_ENABLE_GST_VIDEOSTREAMING. As long as you have gstreamer installed there shouldn't be an issue, unless you just don't want it

omid-esrafilian, 2024-12-09:
@HTRamsey 
Thanks so much for your reply. 
I disabled it, and it works now. Perhaps since GStreamer is optional to install, QGC_ENABLE_GST_VIDEOSTREAMING can be disabled by default to avoid getting such an error.

HTRamsey, 2024-12-09:
Only issue with that is I think most people expect video to be included by default. Maybe we can do that once the QtMultimedia video receiver is working more reliably

new-user-name, 2025-02-24:
It seems to be related with this part of the CMakeLists.txt file located in 
C:\[path]\qgroundcontrol\src\VideoManager\VideoReceiver\GStreamer\gstqml6gl

```
if(MACOS)
    # Using FindGStreamer.cmake is currently bypassed on MACOS since it doesn't work
    # So for now we hack in a simple hardwired setup which does work
    find_library(GSTREAMER_FRAMEWORK GStreamer)
    set(GST_PLUGINS_VERSION 1.24.12)
    set(GSTREAMER_FRAMEWORK_PATH "/Library/Frameworks/GStreamer.framework")
    target_link_libraries(gstqml6gl PUBLIC "$<LINK_LIBRARY:FRAMEWORK,${GSTREAMER_FRAMEWORK_PATH}>")
    target_include_directories(gstqml6gl PUBLIC "${GSTREAMER_FRAMEWORK_PATH}/Headers")
else()
    set(GST_COMPONENTS GlEgl GlPrototypes GlWayland GlX11)
    find_package(GStreamer REQUIRED COMPONENTS OPTIONAL_COMPONENTS ${GST_COMPONENTS})
    target_link_libraries(gstqml6gl PUBLIC GStreamer::GStreamer)
endif()
```
the else part fires on Windows, and the error is on the line
`find_package(GStreamer REQUIRED COMPONENTS OPTIONAL_COMPONENTS ${GST_COMPONENTS})`

HTRamsey, 2025-02-24:
@new-user-name what error are you getting? As far as I can tell the gstreamer building has been fixed on most/all supported systems.

new-user-name, 2025-02-24:
> [[image]@new-user-name](https://github.com/new-user-name?rgh-link-date=2025-02-24T15%3A28%3A28.000Z) what error are you getting? As far as I can tell the gstreamer building has been fixed on most/all supported systems.

```
Could NOT find GLIB2 (missing: GLIB2_LIBRARIES GTHREAD2_LIBRARIES
  GLIB2_INCLUDE_DIRS)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  C:/Qt/6.8.2/msvc2022_64/lib/cmake/Qt6/3rdparty/extra-cmake-modules/find-modules/FindGLIB2.cmake:130 (find_package_handle_standard_args)
  C:/Program Files/CMake/share/cmake-3.28/Modules/CMakeFindDependencyMacro.cmake:76 (find_package)
  C:/Qt/6.8.2/msvc2022_64/lib/cmake/Qt6/FindGObject.cmake:20 (find_dependency)
  C:/Program Files/CMake/share/cmake-3.28/Modules/CMakeFindDependencyMacro.cmake:76 (find_package)
  cmake/find-modules/FindGStreamer.cmake:125 (find_dependency)
  src/VideoManager/VideoReceiver/GStreamer/gstqml6gl/CMakeLists.txt:15 (find_package)
```

HTRamsey, 2025-02-24:
Are you on the latest build? Previously Pkgconfig wasn't found properly on windows but it is now and should be working fine. Either that or you might be on windows Arm or something.

new-user-name, 2025-02-24:
> Are you on the latest build?

Yep. What I have to do?
`choco install pkgconfiglite` ?

HTRamsey, 2025-02-24:
No, it finds it in the installed gstreamer folder. The only other guess I have is that your gstreamer isn't instead correctly.

new-user-name, 2025-02-24:
> No, it finds it in the installed gstreamer folder. The only other guess I have is that your gstreamer isn't instead correctly.

You are absolutely right!.. I downloaded Gstreamer, but did not install it. What a shame.

closed 2025-02-24 by HTRamsey