mavlink/qgroundcontrol #12186: Can't build master branch with qt creator

DieBorr, 2024-12-09:
After 1fe58a934fe4dd2f962d86377ab2593bc56714e2 made by @HTRamsey  I'm not able to build qgc, as I'm not an expert with Cmake I don't know if it's my fault or is just a bug, here is a screenshot:

[image]

### Here is the CmakeOutput:
---------------------------
CMake Error in src/VideoManager/VideoReceiver/GStreamer/gstqml6gl/CMakeLists.txt:
  Imported target "GStreamer::GStreamer" includes non-existent path
 
/usr/include/graphene-1.0

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

-----------------------------
Thanks!

DieBorr, 2024-12-10:
I forgot to say, I was trying to build in Ubuntu 22.04 (jammy Jellyfish)

omid-esrafilian, 2024-12-10:
I think this is similar to the issue (#12182) I encountered. If you don't need GStreamer, you need to disable the "QGC_ENABLE_GST_VIDEOSTREAMING" option in cmake->CustomOptions.cmake file as follows:

option(QGC_ENABLE_GST_VIDEOSTREAMING "Enable GStreamer Video Backend" OFF)

After that, also delete the build directory and the "CMakeLists.txt.user" file, then import your project again into Qt Creator, and try to build the project. 

I hope that helps.

DieBorr, 2024-12-10:
> I think this is similar to the issue (#12182) I encountered. If you don't need GStreamer, you need to disable the "QGC_ENABLE_GST_VIDEOSTREAMING" option in cmake->CustomOptions.cmake file as [...]

Thanks, right now I won't use GStreamer but I would like to use it soon, that's why I'm asking if it's my problem or just a bug

jorgitinlugo, 2024-12-10:
@DieBorr did you get this error?

[image]

HTRamsey, 2024-12-10:
@DieBorr you should be good now. I forgot about how linux has graphene in a different location than every other platform

closed 2024-12-10 by HTRamsey

DieBorr, 2024-12-10:
> @DieBorr did you get this error?
> 
> [image]
> 

No, just the message of the issue ticket.
As @HTRamsey said it should be fixed now, so if you keep having that problem feel free to open another one and we would try to help you.
Thanks!