mavlink/qgroundcontrol #12211: [MSVC]Build failed after updating the master branch to the latest commit

brianGriifin114, 2024-12-16:
We are the MSVC backend test team and use QgroundControl as part of the open source project testing. Recently, after we updated the latest commit, QGC build failed at 
```
"The build of 'C:\gitP\mavlink\qgroundcontrol\build_amd64\CMakeFiles\d691135841483b600836294878b4cac8\Joystick_qmltyperegistration.rule' depends on 'C:\GITP\MAVLINK\QGROUNDCONTROL\BUILD_AMD64\SRC\JOYSTICK\JOYSTICK_QMLTYPEREGISTRATIONS.CPP' which is produced by the build of 'C:\gitP\mavlink\qgroundcontrol\build_amd64\CMakeFiles\b838792b63b78775530783403db177af\joystick_qmltyperegistrations.cpp.rule'. The items cannot be built in parallel."
C:\Program Files\Microsoft Visual Studio\2022\Enterprise\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(254,5): error MSB8066: Custom build for 'C:\gitP\mavlink\qgroundcontrol\build_amd64\CMakeFiles\b838792b63b78775530783403db177af\joystick_qmltyperegistrations.cpp.rule;C:\gitP\mavlink\qgroundcon trol\build_amd64\CMakeFiles\d691135841483b600836294878b4cac8\Joystick_qmltyperegistration.rule;C:\gitP\mavlink\qgroundcontrol\src\Joystick\CMakeLists.txt' exited with code 1. [C:\gitP\mavlink\qgroundcontrol\build_amd64\src\Joystick\Joystick_qmltyperegistration.vcxproj]
```
The build system detected that joystick_qmltyperegistrations.cpp is a dependent file, but it is also the output of another custom rule.
Two tasks are trying to access or generate the same file in parallel, causing a conflict.
Have you ever encountered such a problem? Thanks! I attached the build.log for reference.
[Build.log](https://github.com/user-attachments/files/18147607/Build.log)

HTRamsey, 2024-12-16:
Have you tried building via QtCreator? I think there's some issue with the QGroundControl.rc & resource.h files but I don't use VC so I'm not entirely sure

brianGriifin114, 2024-12-17:
I'll try disabling parallel builds. As for building through QTCreator, we need to build with msbuild to detect if there are any problems with the compiler, so this build method is not helpful for us😅. Thank you for your attention!

HTRamsey, 2024-12-17:
Ah okay gotcha. The documentation is made for building via QtCreator so you'd have to resolve VS on your own. Let me know if you figure it out

brianGriifin114, 2024-12-17:
Sure thing🫡

brianGriifin114, 2024-12-17:
Disabling parallel builds allows the build to pass without a significant increase in build time.

closed 2024-12-17 by brianGriifin114