mavlink/qgroundcontrol #12237: ninja: error: dependency cycle

DroneerUAV, 2024-12-19:
I try to build the master branch and I get this error,How can I solve this problem?

ninja: error: dependency cycle: cmake_object_order_depends_target_APMFirmwarePlugin -> cmake_object_order_depends_target_PX4FirmwarePlugin -> cmake_object_order_depends_target_FollowMe -> cmake_object_order_depends_target_Geo -> cmake_object_order_depends_target_Gimbal -> cmake_object_order_depends_target_GPS -> cmake_object_order_depends_target_Joystick -> qml/QGroundControl/JoystickManager/Joystick.qmltypes -> src/FirmwarePlugin/meta_types/qt6firmwareplugin_debug_metatypes.json -> src/FirmwarePlugin/meta_types/FirmwarePlugin_json_file_list.txt -> src/FirmwarePlugin/APM/APMFirmwarePlugin.lib -> src/FirmwarePlugin/APM/CMakeFiles/APMFirmwarePlugin.dir/APMFirmwarePlugin_autogen/mocs_compilation.cpp.obj -> cmake_object_order_depends_target_APMFirmwarePlugin

HTRamsey, 2024-12-19:
I know this happens on Qt6.8, which is why I'm probably going to remove the individual cmake libs. But you shouldn't be having a problem on Qt6.6 as far as I know. At least I don't think anyone else has

DroneerUAV, 2024-12-20:
> I know this happens on Qt6.8, which is why I'm probably going to remove the individual cmake libs. But you shouldn't be having a problem on Qt6.6 as far as I know. At least I don't think anyone else has

Yes, the build kit I use is Qt6.8, I will try to use Qt6.6 to build it, thanks

HTRamsey, 2024-12-20:
Yup, will resolve the Qt6.8 issue and update to that before the next release

closed 2024-12-20 by HTRamsey