mavlink/qgroundcontrol #12239: Not possible to select the Spanish language

copilador, 2024-12-20:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/support/support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Being able to select the language es_ES in the configuration under Miscellaneous -> Language in QGroundControl.

## Current Behavior
Several languages appear, but not Spanish (es_ES).

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Open the application ./QGroundControl.AppImage
2 .Select Tool -> Application Settings -> Miscellaneous -> Language

## System Information
When posting bug reports, include the following information
- 138-Ubuntu SMP Sat Nov 30 22:28:23 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

./QGroundControl.AppImage --version
Settings location "/home/joan/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(Spanish, Latin, Spain) ; Name "es_ES" ; Preffered (used in maps):  "es-ES"
LocalizationLog: Qt lib localization for "es_ES" is not present
VideoReceiverLog: Stop called on empty URI
VideoReceiverLog: Stop called on empty URI
MAVLinkLogManagerLog: MAVLink logs directory: "/home/joan/Documentos/QGroundControl/Logs"
qt.network.ssl: QSslSocket: cannot resolve EVP_PKEY_base_id
qt.network.ssl: QSslSocket: cannot resolve SSL_get_peer_certificate
Ignoring ssl certificates due to OpenSSL version mismatch
Map Cache in: "/home/joan/.cache/QGCMapCache300" / "qgcMapCache.db"
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
qt.network.ssl: QSslSocket: cannot call unresolved function EVP_PKEY_base_id
"v4.4.3"

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/settings_view/console_logging.html)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.
[image]

HTRamsey, 2024-12-21:
Only finished (or nearly finished) translations are included in the releases I believe. Somebody would have to provide translations through Crowdin for it to be in a release

copilador, 2024-12-21:
Thank you for the clarification, I was unaware that the translation status was not finished.

closed 2024-12-21 by copilador

DonLakeFlyer, 2024-12-21:
Correct. Daily builds have all languages. Even if incomplete.

copilador, 2024-12-21:
OK, I have downloaded the Daily builds, and it does allow the selection of the Spanish language. Thanks again!