mavlink/qgroundcontrol #12245: Remote ID Toolbar bitmap problems

DonLakeFlyer, 2024-12-21:
Doesn't work in the light palette since the "ID" portion of the image is white.

DonLakeFlyer, 2025-02-11:
@Davidsastresas Do you have anyone to work on the remote id toolbar icon?

Davidsastresas, 2025-02-12:
Sure! @alexdelatorre can help.

Lately we are doing a different approach on those kind of icons, and it is proving to be nice to handle.

If the icon design is fine and we want to maintain it, I would propose a single icon, .svg, with the background of the "little body" part of the icon, so we can color it from code. And then we manage manually the qml indicator to overlay a QGCLabel, so hopefully it adjusts automatically to light palettes.

I think we can manage from here, I will keep you posted!

closed 2025-02-14 by DonLakeFlyer