mavlink/qgroundcontrol #12256: how can i update serial driver to solve none serialport available i want to all steps , please

Yahea213, 2024-12-25:
Please only ask questions related to source code.
- For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources
- If your question still isn't answered, please check the forums: https://discuss.px4.io/c/qgroundcontrol/15.
- If it is about Qt or any QGC dependencies, please refer to them instead.

HTRamsey, 2025-01-01:
Need more info to help.

Yahea213, 2025-01-01:
i Built qgroundcontrol4.4.2 with qt5 but the serial port none available and they said about solve this problem(serial port none available) by update the serial driver but the updating to this library(usb-serial-for-android) does not compatible with qt5 
there are solve

HTRamsey, 2025-01-02:
It's updated in the daily build. Otherwise it is quite a bit of work to get it to work for v4.4

Yahea213, 2025-01-04:
yes, i think this 
thank you

closed 2025-01-04 by HTRamsey