mavlink/qgroundcontrol #12260: Qgroundcontrol hangs on mid-2010 MacBook Pro running 10.13.6 - High Sierra

joe-stroup, 2024-12-28:
According to the Qgroundcontrol download page, MacOS is support with MacOS 10.11 or later.

Reference: https://docs.qgroundcontrol.com/master/en/qgc-user-guide/getting_started/download_and_install.html

When I try running Qgroundcontrol, it starts, but hangs up after it asks for permission to use location services, but before it finishes painting the map on the screen.

It may be that a newer version of MacOS (or a newer Mac) are required.  But if I'm doing something wrong - or there is a bug - I thought I should make this report.

I'm happy to test anything assist the DEVs.

Thanks!

DonLakeFlyer, 2024-12-28:
I think Stable is >=10.13 and daily is >12. I'll update the docs.

joe-stroup, 2024-12-28:
Thanks for the quick (and helpful) response.  How do I make sure I'm downloading the "Stable" version?  (I did downloads from two different web pages when I googled "qgroundcontrol download install".)  Once I download "Stable" and run it, I'll report on how it works.  Once again - Thanks!

HTRamsey, 2025-01-01:
You can tell from the settings popup which version you are using like in this screenshot. Unless where you download mentions "Daily Build" then it's probably a stable.
[image]

daijo, 2025-01-01:
Have had the same issue for months. Best way to tell which version you use is that stable freezes (before drawing the map) and daily just crashes (gstreamer). Using a Windows laptop exclusively to run QGroundcontrol.

HTRamsey, 2025-01-02:
Probably the best option right now for people using macos would be to disable video on the daily and build manually until the gstreamer linking is fixed. Or you might be able to use the QtMultimedia receiver instead through the custom cmake options, not sure if it'll work though as it's largely untested on most platforms.

oliver408i, 2025-01-03:
I have a macbook pro m2 on 15.3 dev beta and I can use qgc without any issue, although it does occasionally crash when removing usb radios, but I haven't had any issues in flight yet.

DonLakeFlyer, 2025-01-03:
> So if there's enough justification to fix this bug, I'd be really very grateful.

It's not a bug. It's the support provided by the Qt library which is used by QGC. We can only support what they support. And as we create new versions of QGC we move forward with newer version of Qt which in turn brings along new OS requirements.

Video problems with daily are a totally separate things from supported OS versions.

closed 2025-01-14 by HTRamsey