mavlink/qgroundcontrol #12261: qgroundcontrol.pro - why !VideoEnabled  ?

sohrab-niche, 2024-12-29:
<pre>
Please only ask questions related to source code.
- For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources
- If your question still isn't answered, please check the forums: https://discuss.px4.io/c/qgroundcontrol/15.
- If it is about Qt or any QGC dependencies, please refer to them instead.
</pre>

Please see
https://raw.githubusercontent.com/mavlink/qgroundcontrol/refs/heads/Stable_V4.4/qgroundcontrol.pro

Why is there '!' in the line below? Shouldn't the '!' sign NOT be there?
```
!VideoEnabled {
    INCLUDEPATH += \
        src/VideoReceiver

    HEADERS += \
        src/VideoManager/GLVideoItemStub.h \
        src/VideoReceiver/VideoReceiver.h

    SOURCES += \
        src/VideoManager/GLVideoItemStub.cc
}
```

HTRamsey, 2025-01-02:
Nah, those files are only used when there is no video.

closed 2025-01-02 by HTRamsey

sohrab-niche, 2025-01-05:
Thank you for the reply and sorry for the late follow-up @HTRamsey. 

Can you please explain what exactly you mean by 'when there is no video'?

Basically, when I tried to build QGroundControl with gstreamer support, I found that without removing the '!', it doesn't build. I'm on a custom version of 4.2.6 but this code snippet has not changed.

I think I'm doing something unexpected or not doing something and so I need to remove the '!' for my build to succeed. Hence wanted to check.

HTRamsey, 2025-01-05:
The only purpose of that section with `!VideoEnabled` is to include GLVideoItemStub.h which is literally just an empty class to replace the implemented VideoReceiver that is not included. If you want video enabled then GLVideoItemStub.h does not need to be included