mavlink/qgroundcontrol #12266: VTOL Landing Pattern gets split into multiple waypoints

anaam-wingxpand, 2024-12-30:
## Expected Behavior
VTOL Landing Pattern should always display as one waypoint called "Landing Pattern" in the mission planning editor.

## Current Behavior
VTOL Landing Pattern only displays as one waypoint when you create a new mission and upload it.
If you later download that same mission, it displays as "Land start", "Camera trigger distance", "Stop image capture", "Stop video capture", "Loiter (altitude)", and "VTOL land".

Before:
[image]

After:
[image]

This is very confusing to the user. 

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Upload a VTOL mission with a landing pattern
2. Download that mission by using File > Download, or by reopening QGC

## System Information
When posting bug reports, include the following information
- Operating System: Windows, Linux, Android
- QGC Version: 4.4.3
- QGC build: Stable
- Flight Controller: SITL, Cubepilot Cube Orange +
- Autopilot (with version): PX4 v1.15

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/settings_view/console_logging.html)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

DonLakeFlyer, 2024-12-30:
Hopefully fixed in #12238

anaam-wingxpand, 2024-12-30:
i fixed it by making the VTOL isValidLandItem function check for MAV_CMD_NAV_VTOL_LAND, instead of MAV_CMD_NAV_LAND.

bool VTOLLandingComplexItem::_isValidLandItem(const MissionItem& missionItem) { if (missionItem.command() != MAV_CMD_NAV_VTOL_LAND ||

anaam-wingxpand, 2024-12-30:
i fixed it by making the VTOL isValidLandItem function check for MAV_CMD_NAV_VTOL_LAND, instead of MAV_CMD_NAV_LAND.

`bool VTOLLandingComplexItem::_isValidLandItem(const MissionItem& missionItem) { if (missionItem.command() != MAV_CMD_NAV_VTOL_LAND ||`

anaam-wingxpand, 2024-12-30:
> i fixed it by making the VTOL isValidLandItem function check for MAV_CMD_NAV_VTOL_LAND, instead of MAV_CMD_NAV_LAND.
> 
> `bool VTOLLandingComplexItem::_isValidLandItem(const MissionItem& missionItem) { if (missionItem.command() != MAV_CMD_NAV_VTOL_LAND ||`

closed 2024-12-30 by anaam-wingxpand