mavlink/qgroundcontrol #12273: Windows 11 black screen

NotAlessandroBeatini, 2025-01-03:
After succesfully installing Qgroundcontrol using the installer, if i launch it the following happens:
- If I launch the "normal" .exe, the window will open and for half a second I will see the UI, then everything will go black and from the task manager QCG will be 0% CPU 
- the other two (GPU safe and compatibility mode) will keep the CPU busy indefinetely but will not make any window appear.

## System Information
When posting bug reports, include the following information
- Operating System: Windows 11 
- QGC Version: not sure, i used the installer, i guess the latest

HTRamsey, 2025-01-03:
Where did you get the installer? Need to know if it's the latest release or daily build

NotAlessandroBeatini, 2025-01-05:
Hi, from here : https://docs.qgroundcontrol.com/master/en/qgc-user-guide/getting_started/download_and_install.html#windows

HTRamsey, 2025-01-05:
Ah then it's an old release and I'm not surprised it isn't working

NotAlessandroBeatini, 2025-01-05:
oh I see ! Is there an installer for the new releases?

HTRamsey, 2025-01-05:
You can get daily builds here: https://docs.qgroundcontrol.com/master/en/qgc-user-guide/releases/daily_builds.html

closed 2025-05-03 by HTRamsey