mavlink/qgroundcontrol #12279: QGC crashing on exit

DonLakeFlyer, 2025-01-04:
I get this:
`ASSERT failure in PlainType *QtGlobalStatic::ApplicationHolder<(anonymous namespace)::Q_QAS__qgcCorePluginInstance>::pointer() [QAS = (anonymous namespace)::Q_QAS__qgcCorePluginInstance]: "The application static was used without a QCoreApplication instance", file /Users/don/Qt/6.7.3/macos/lib/QtCore.framework/Headers/qapplicationstatic.h, line 55`

From here:
```
VideoManager::~VideoManager()
{
    for (VideoReceiverData &videoReceiver : _videoReceiverData) {
        QGCCorePlugin::instance()->releaseVideoSink(videoReceiver.sink);
        delete videoReceiver.receiver;
        videoReceiver.receiver = nullptr;
    }

    // qCDebug(VideoManagerLog) << Q_FUNC_INFO << this;
}
```

Stack:
[image]

HTRamsey, 2025-01-04:
Oh you're doing this with video disabled right?

DonLakeFlyer, 2025-01-04:
Oh, wait. Crap, I'm running 6.7.3. That might be the problem.

DonLakeFlyer, 2025-01-04:
This is with video enabled, but only happens on 6.7.3

PR #12145 (2025-01-07): Update to Qt 6.8
This is an LTS and will be the version used for v5
Note: Qt6.7 is unusable for Android

closes #12279
closed 2025-01-07 by HTRamsey