mavlink/qgroundcontrol #12314: RTCM INJECTION BUT NO RTK STATUS/STATUS

jorgitinlugo, 2025-01-10:
Hello,

I’m working on a project involving an RTK system. In my setup, the base antenna is connected to an Odroid C4, and the rover antenna is mounted on the drone. I’ve written a script that sends RTCM corrections from the Odroid to the drone.

Everything seems to be working as expected:

When I check the drone in QGroundControl and run the command gps status in the terminal, I can see that RTCM injection is happening.
However, the GPS fix type on the drone does not change to 5 (RTK float) or 6 (RTK fixed).

[image]

What could be causing this issue? Is there something I’m missing in the configuration or setup that prevents the GPS from achieving an RTK fix?

Any guidance would be greatly appreciated.

Thanks in advance!

jorgitinlugo, 2025-01-10:
And if i try to send it to the Qgroundcontrol by using Comm Links, i dont get rtcm_injection using gps_status but i get this in the Mavlink inspector:

[image]

The problem is the same, i dont get fix type correctly.

DonLakeFlyer, 2025-01-10:
There is no support for injecting base station RTK in QGC. Only direct connection to an RTK device over usb is supported.

jorgitinlugo, 2025-01-13:
@DonLakeFlyer but i dont want Qground to send RTCM corrections, RTCM corrections are sent directly from the Odroid to the drone, the question i have is: With this scheme, I am able to see in qground a 5/6 gps fix type?