mavlink/qgroundcontrol #12315: build QGC android master QT6.8.1 error

lequann, 2025-01-11:
[image]

INFOR :
[image]

HTRamsey, 2025-01-11:
You could try changing the git tag on [this line](https://github.com/mavlink/qgroundcontrol/blob/e0fd8eb111f2c6d9ccdc1b6663d5ee6ef8b8b748/src/AnalyzeView/CMakeLists.txt#L172) to 'main' instead. I think they've had some recent commits fixing finding dependencies

HTRamsey, 2025-01-11:
Their docs also mention this:
[image]

lequann, 2025-01-12:
Thank you guy! I will try it ...

Daksh7112003, 2025-01-21:
Hey lequann can u help me regarding the setup for android ,
I m getting streamer errors

[image]

lequann, 2025-01-23:
Hi Daksh7112003.
I have built the apk file.
However the app crashes when starting up. I am looking for the cause to fix it

closed 2025-01-23 by lequann