mavlink/qgroundcontrol #12323: Automation for MissionPlan in QGC

tranhien1612, 2025-01-15:
I want to do automatic flight from A(waypoint) to B, C, D (waypoint).. .Currently I find myself having to manually enter GPS information, aircraft information for those points one by one. I'm trying to simplify that process.
Suppose there is a button to start creating a mission, then the user will control the plane to fly to points A, B, C, D. When reaching those points, there is an additional button to save information at those locations. When completed, an automatic flight plan will be created with the information of points A, B, C, D taken before. Suppose point A is takeoff, point B is at GPS 1 coordinates with lat/lon information, angle of aircraft, ....
I think that would simplify the usage instead of the user having to enter the waypoints information (in case there are too many waypoints).

DonLakeFlyer, 2025-01-15:
* Fly the vehicle to where you want the waypoint to be
* Add a new waypoint
* Click the hamburger to drop the edit menu, select Edit Position
* [image]
* Pick Vehicle Position
* [image]
* Click Set Position / Move
* You waypoint is now at the vehicle position

closed 2025-01-15 by DonLakeFlyer

tranhien1612, 2025-01-16:
@DonLakeFlyer Can I get other information like plane direction, plane speed, gimbal angle... like you said?

DonLakeFlyer, 2025-01-16:
Only lat/lon/alt is supported.