mavlink/qgroundcontrol #12324: Built from source, the app is opening and closing repeatedly

hseysen, 2025-01-15:
## System Information
- Operating System: Windows 11
- QT: 6.8.1
- QGC build: daily, release

## Description
I cloned the repository, didn't make major changes (just experimented by changing some colors). Built using `Desktop Qt 6.8.1 MSVC2022 64bit`. Runs perfectly fine through QTCreator IDE. Then I opened `Developer Command Prompt for VS 2022` and ran `cmake --install . --config Release`. Process required me to install `makensis.exe`, so I installed from SourceForge. At the end, I got my `QGroundControl-installer.exe` file, and it ran, installed everything without complaints. But when I open `%{install_dir}/bin/QGroundControl.exe`, I get this empty window opening and closing repeatedly until I close it from Task Manager. I've followed the instructions on [the documentation](https://docs.qgroundcontrol.com/master/en/qgc-dev-guide/getting_started/index.html), but I'm not sure if I'm missing anything.

## Shortened logs from build process:
```pwsh
> cmake --install . --config Release
-- Writing C:/Users/Hesen/Desktop/qgroundcontrol/build/Desktop_Qt_6_8_1_MSVC2022_64bit-Release/staging/bin/qt.conf
-- Running Qt deploy tool for C:/Users/Hesen/Desktop/qgroundcontrol/build/Desktop_Qt_6_8_1_MSVC2022_64bit-Release/QGroundControl.exe in working directory 'C:/Users/Hesen/Desktop/qgroundcontrol/build/Desktop_Qt_6_8_1_MSVC2022_64bit-Release/staging'
'C:/Qt/6.8.1/msvc2022_64/bin/windeployqt.exe' 'C:/Users/Hesen/Desktop/qgroundcontrol/build/Desktop_Qt_6_8_1_MSVC2022_64bit-Release/QGroundControl.exe' '--dir' '.' '--libdir' 'bin' '--plugindir' 'plugins' '--qml-deploy-dir' 'qml' '--translationdir' 'translations' '--force' '--qtpaths' 'C:/Qt/6.8.1/msvc2022_64/bin/qtpaths6.exe' 'qml/QtQuick/qtquick2plugin.dll' 'qml/QtQml/qmlplugin.dll' 'qml/QtQml/Models/modelsplugin.dll' 'qml/QtQml/WorkerScript/workerscriptplugin.dll' 'qml/QtQuick/Controls/qtquickcontrols2plugin.dll' 'qml/QtQuick/Controls/Fusion/qtquickcontrols2fusionstyleplugin.dll' 'qml/QtQuick/Controls/Material/qtquickcontrols2materialstyleplugin.dll' 'qml/QtQuick/Controls/Imagine/qtquickcontrols2imaginestyleplugin.dll' 'qml/QtQuick/Controls/Universal/qtquickcontrols2universalstyleplugin.dll' 'qml/QtQuick/Controls/FluentWinUI3/qtquickcontrols2fluentwinui3styleplugin.dll' 'qml/QtQuick/Controls/Windows/qtquickcontrols2windowsstyleplugin.dll' 'qml/QtQuick/Controls/Basic/qtquickcontrols2basicstyleplugin.dll' 'qml/QtQuick/Templates/qtquicktemplates2plugin.dll' 'qml/QtQuick/Controls/impl/qtquickcontrols2implplugin.dll' 'qml/QtQuick/Controls/Fusion/impl/qtquickcontrols2fusionstyleimplplugin.dll' 'qml/QtQuick/Window/quickwindowplugin.dll' 'qml/QtQuick/Controls/Material/impl/qtquickcontrols2materialstyleimplplugin.dll' 'qml/QtQuick/Controls/Imagine/impl/qtquickcontrols2imaginestyleimplplugin.dll' 'qml/QtQuick/Controls/Universal/impl/qtquickcontrols2universalstyleimplplugin.dll' 'qml/QtQuick/Controls/FluentWinUI3/impl/qtquickcontrols2fluentwinui3styleimplplugin.dll' 'qml/QtQuick/Effects/effectsplugin.dll' 'qml/QtQuick/Layouts/qquicklayoutsplugin.dll' 'qml/QtQuick/Shapes/qmlshapesplugin.dll' 'qml/QtQuick/NativeStyle/qtquickcontrols2nativestyleplugin.dll' 'qml/QtQuick/Controls/Windows/impl/qtquickcontrols2windowsstyleimplplugin.dll' 'qml/QtQuick/Controls/Basic/impl/qtquickcontrols2basicstyleimplplugin.dll' 'qml/QtQuick/Dialogs/qtquickdialogsplugin.dll' 'qml/QtQuick/Dialogs/quickimpl/qtquickdialogs2quickimplplugin.dll' 'qml/Qt/labs/folderlistmodel/qmlfolderlistmodelplugin.dll' 'qml/Qt5Compat/GraphicalEffects/qtgraphicaleffectsplugin.dll' 'qml/Qt5Compat/GraphicalEffects/private/qtgraphicaleffectsprivateplugin.dll' 'qml/Qt/labs/qmlmodels/labsmodelsplugin.dll' 'qml/QtCharts/qtchartsqml2plugin.dll' 'qml/QtMultimedia/quickmultimediaplugin.dll' 'qml/QtLocation/declarative_locationplugin.dll' 'qml/QtPositioning/positioningquickplugin.dll' 'qml/Qt/labs/animation/labsanimationplugin.dll' 'qml/Qt/labs/platform/labsplatformplugin.dll' 'qml/QtCore/qtqmlcoreplugin.dll' 'qml/QtQuick3D/qquick3dplugin.dll'
C:\Users\Hesen\Desktop\qgroundcontrol\build\Desktop_Qt_6_8_1_MSVC2022_64bit-Release\QGroundControl.exe 64 bit, release executable [QML]
Adding in plugin type generic for module: Qt6Gui
Adding in plugin type geoservices for module: Qt6Location
Adding in plugin type iconengines for module: Qt6Gui
Adding Qt6Svg for qsvgicon.dll from plugin type: iconengines
Adding in plugin type imageformats for module: Qt6Gui
Adding in plugin type multimedia for module: Qt6Multimedia
Adding in plugin type networkinformation for module: Qt6Network
Adding in plugin type platforminputcontexts for module: Qt6Gui
Adding Qt6VirtualKeyboard for qtvirtualkeyboardplugin.dll from plugin type: platforminputcontexts
Adding in plugin type platforms for module: Qt6Gui
Adding in plugin type position for module: Qt6Positioning
Adding in plugin type qmltooling for module: Qt6Qml
Adding in plugin type sensors for module: Qt6Sensors
Adding in plugin type sqldrivers for module: Qt6Sql
Adding in plugin type styles for module: Qt6Widgets
Adding in plugin type texttospeech for module: Qt6TextToSpeech
Adding in plugin type tls for module: Qt6Network
Skipping plugin qopensslbackend.dll. Use -force-openssl or specify -openssl-root if you want to use it.
Direct dependencies: Qt6Bluetooth Qt6Charts Qt6ChartsQml Qt6Core Qt6Gui Qt6LabsAnimation Qt6LabsFolderListModel Qt6LabsPlatform Qt6LabsQmlModels Qt6Location Qt6Multimedia Qt6MultimediaQuick Qt6Network Qt6OpenGL Qt6OpenGLWidgets Qt6Positioning Qt6PositioningQuick Qt6Qml Qt6QmlCore Qt6QmlMeta Qt6QmlModels Qt6QmlWorkerScript Qt6Quick Qt6Quick3D Qt6Quick3DRuntimeRender Qt6Quick3DUtils Qt6QuickControls2 Qt6QuickControls2Basic Qt6QuickControls2BasicStyleImpl Qt6QuickControls2FluentWinUI3StyleImpl Qt6QuickControls2Fusion Qt6QuickControls2FusionStyleImpl Qt6QuickControls2Imagine Qt6QuickControls2ImagineStyleImpl Qt6QuickControls2Impl Qt6QuickControls2Material Qt6QuickControls2MaterialStyleImpl Qt6QuickControls2Universal Qt6QuickControls2UniversalStyleImpl Qt6QuickControls2WindowsStyleImpl Qt6QuickDialogs2 Qt6QuickDialogs2QuickImpl Qt6QuickDialogs2Utils Qt6QuickEffects Qt6QuickLayouts Qt6QuickShapes Qt6QuickTemplates2 Qt6Sensors Qt6SerialPort Qt6ShaderTools Qt6Sql Qt6TextToSpeech Qt6Widgets Qt6Xml
All dependencies   : Qt6Bluetooth Qt6Charts Qt6ChartsQml Qt6Core Qt6Gui Qt6LabsAnimation Qt6LabsFolderListModel Qt6LabsPlatform Qt6LabsQmlModels Qt6Location Qt6Multimedia Qt6MultimediaQuick Qt6Network Qt6OpenGL Qt6OpenGLWidgets Qt6Positioning Qt6PositioningQuick Qt6Qml Qt6QmlCore Qt6QmlMeta Qt6QmlModels Qt6QmlWorkerScript Qt6Quick Qt6Quick3D Qt6Quick3DRuntimeRender Qt6Quick3DUtils Qt6QuickControls2 Qt6QuickControls2Basic Qt6QuickControls2BasicStyleImpl Qt6QuickControls2FluentWinUI3StyleImpl Qt6QuickControls2Fusion Qt6QuickControls2FusionStyleImpl Qt6QuickControls2Imagine Qt6QuickControls2ImagineStyleImpl Qt6QuickControls2Impl Qt6QuickControls2Material Qt6QuickControls2MaterialStyleImpl Qt6QuickControls2Universal Qt6QuickControls2UniversalStyleImpl Qt6QuickControls2WindowsStyleImpl Qt6QuickDialogs2 Qt6QuickDialogs2QuickImpl Qt6QuickDialogs2Utils Qt6QuickEffects Qt6QuickLayouts Qt6QuickShapes Qt6QuickTemplates2 Qt6Sensors Qt6SerialPort Qt6ShaderTools Qt6Sql Qt6TextToSpeech Qt6Widgets Qt6Xml
To be deployed     : Qt6Bluetooth Qt6Charts Qt6ChartsQml Qt6Core Qt6Gui Qt6LabsAnimation Qt6LabsFolderListModel Qt6LabsPlatform Qt6LabsQmlModels Qt6Location Qt6Multimedia Qt6MultimediaQuick Qt6Network Qt6OpenGL Qt6OpenGLWidgets Qt6Positioning Qt6PositioningQuick Qt6Qml Qt6QmlCore Qt6QmlMeta Qt6QmlModels Qt6QmlWorkerScript Qt6Quick Qt6Quick3D Qt6Quick3DRuntimeRender Qt6Quick3DUtils Qt6QuickControls2 Qt6QuickControls2Basic Qt6QuickControls2BasicStyleImpl Qt6QuickControls2FluentWinUI3StyleImpl Qt6QuickControls2Fusion Qt6QuickControls2FusionStyleImpl Qt6QuickControls2Imagine Qt6QuickControls2ImagineStyleImpl Qt6QuickControls2Impl Qt6QuickControls2Material Qt6QuickControls2MaterialStyleImpl Qt6QuickControls2Universal Qt6QuickControls2UniversalStyleImpl Qt6QuickControls2WindowsStyleImpl Qt6QuickDialogs2 Qt6QuickDialogs2QuickImpl Qt6QuickDialogs2Utils Qt6QuickEffects Qt6QuickLayouts Qt6QuickShapes Qt6QuickTemplates2 Qt6Sensors Qt6SerialPort Qt6ShaderTools Qt6Sql Qt6Svg Qt6TextToSpeech Qt6VirtualKeyboard Qt6Widgets Qt6Xml
Updating dlls...
Creating qms....
-- Creating Win Installer
[... cut ...]
Uninstall: 2 pages (192 bytes), 1 section (2072 bytes), 205 instructions (5740 bytes), 120 strings (3366 bytes), 1 language table (254 bytes).

Using lzma (compress whole) compression.

EXE header size:               89600 / 38912 bytes
Install code:                          (67588 bytes)
Install data:                          (503773760 bytes)
Uninstall code+data:                   (97852 bytes)
Compressed data:           146782446 / 503939200 bytes
CRC (0xD936ABA0):                  4 / 4 bytes

Total size:                146872050 / 503978116 bytes (29.1%)" ; NSIS_ERROR=""
```

hseysen, 2025-01-16:
Additionally, I don't know if this is related to the issue or not, the docs refer to [_MSVC 2019 64 bit_](https://docs.qgroundcontrol.com/master/en/qgc-dev-guide/getting_started/index.html#install-qt), but QT 6.8.1 does not offer this option. The available option is MSVC 2022.

[image]

DonLakeFlyer, 2025-02-12:
I'm thinking this was maybe the exiv2 dll problem. That was fixed in the last week or so. You should try again with latest master.

DonLakeFlyer, 2025-02-17:
> I cannot get any console message to be able to identify the issue.

Did you run it from the command line to see if you get console output there?

vryabokon1705, 2025-02-18:
> Did you run it from the command line to see if you get console output there?
Yes, i run from command line and do not have any output

[image]

vryabokon1705, 2025-04-14:
In my case, the issue was resolved after switching to the 1.22 gstreamer version

HTRamsey, 2025-04-22:
Yup, 1.24 has an issue with the gstreamer plugin scanner but 1.22 seems to work

closed 2025-04-22 by HTRamsey