mavlink/qgroundcontrol #12325: RTSP not working on daily when compiled : VideoReceiverLog: Failed VideoReceiverLog: gst_element_factory_make('tee') failed

Godeffroy, 2025-01-15:
I'm facing the same problem as [#12133](https://github.com/mavlink/qgroundcontrol/issues/12133), i tried with 4 differents version of gstreamer like 1.18.6 1.20.7 1.22.12 1.24.11 but without success.
I stream a video with vlc with this [doc](https://wiki.videolan.org/Documentation:Streaming_HowTo_New)
The fun fact is the compiled daily version of github is working fine when installed on my computer.

OS: Windows10 19045.5247 + msvc2019 64 bit + msvc2022 64 bit

gstreamer: 1.22.12 gstreamer-1.0-devel-msvc-x86_64-1.22.12.msi + gstreamer-1.0-msvc-x86_64-1.22.12

QT: 6.8.1 + creator 15.0.0

Cmake 3.29.3 (Qt)

[image]

[image]

Beside, to make the compilation working, i had to add few changes to the Qt kit :
First change the compiler C/C++ from msvc2022 (17.12 amd64) to msvc2019 (16.11 amd64) dunno why
Add to path
Path+=D:\Repository\exiv2-0.28.3-2019msvc64\lib
Path+=D:\Repository\exiv2-0.28.3-2019msvc64\include
Path+=D:\Repository\zlib-1.3.1
Path+=D:\Repository\zlib-1.3.1\win32

yizhendongfeng, 2025-03-03:
same problem

HTRamsey, 2025-05-03:
This has likely been resolved, working on improving documentation for people struggling