mavlink/qgroundcontrol #12357: ANDROID GSTREAMER ERROR

Daksh7112003, 2025-01-21:
[image]

These are errors of gstreamer  while doing setup of QGC in android ...

 KINDLY FIX THIS ISSUE 

My confi

[image]

Daksh7112003, 2025-01-21:
I m using master version of QGC and QT6.8.1

HTRamsey, 2025-01-21:
So this is because of an incorrect GStreamer pkgconfig file, will find a solution. This is an example: 
```
prefix=/home/nirbheek/projects/repos/cerbero.git/1.22/build/dist/android_universal/armv7
exec_prefix=${prefix}
libdir=${prefix}/lib
includedir=${prefix}/include

Name: x264
Description: H.264 (MPEG4 AVC) encoder library
Version: 0.157.2M 7552bdb
Libs: -L/home/nirbheek/projects/repos/cerbero.git/1.22/build/dist/android_universal/armv7/lib -lx264 
Libs.private: -lm -ldl
Cflags: -I${prefix}/include
```
The Libs: section is wrong in this one

Daksh7112003, 2025-01-22:
this is the default gstreamer which is downloaded by qgc itself , developers need to fix this issue , 
what is the solution for now ?

HTRamsey, 2025-01-23:
It is still present in newer GStreamers. You can disable GStreamer for now, or if you know how then you can add those libraries to the CMake linking

closed 2025-01-27 by Daksh7112003