mavlink/qgroundcontrol #12358: Available Fly Modes

ezzz01, 2025-01-21:
## Expected Behavior
All Flight modes should be shown in the dropdown, including "Position", "Stabilized" and others. Possibly, the documentation should be improved to explain why/how the available Flight modes should be changed.

## Current Behavior
Only 3 Flight modes are shown in the toolbar dropdown: "Hold", "Mission", "Return".

## System Information
- Operating System: Ubuntu 24.04
- QGC Version: 4.4.3
- QGC build: custom build, tried building both locally on Ubuntu and on Github
- Flight Controller: 
- Autopilot (with version): PX4 1.15.1

## Detailed Description
We have a custom build, based on QGC 4.4.3.
When using our drone and sometimes when using the PX4 simulator, Flight modes dropdown doesn't show the full list of modes, but only 3 of them: "Hold", "Mission", "Return". Please see the screenshots below. They were made with exactly the same simulator, but the apk files were produced once on local system running Ubuntu, and another on Github. 

I suspect that the difference is somehow impacted by the Build Type, so I've tried switching between Daily and Stable, but can't find any pattern, when the Flight mode list changes. I can't find any logic for it in the Github workflow file. I found one commit that introduced logic related to Flight modes based on the build type: https://github.com/Monopulse/UavMap-Android-4.4.3/commit/c002ef6f702be1a0ec52914754ab898e3f0e538b

[image]
[image]

DonLakeFlyer, 2025-01-21:
Does this only show up in your custom build? What happen with a daily build pulled from the web site?

justas-, 2025-01-22:
Just to add - it seems this is purely an UI issue. We can still switch to a required mode via the RC channel mapping and it works as expected.

ezzz01, 2025-01-28:
The problem has been resolved, it turns out we had modified the `CustomFirmwarePlugin` class in the old version and didn't copy it to the new one.

closed 2025-01-28 by ezzz01