mavlink/qgroundcontrol #12364: Error building QGC source on Ubuntu 20.04

hesamira, 2025-01-22:
Hi every one,
I used QGC source, Ubuntu 20.04 and Qt 6.8.1. When I want to build my code, I have faced this error.
Please, help me. Thanks.

nca@nca:~$ qmake --version
QMake version 3.1
Using Qt version 6.8.1 in /usr/local/qt6.8.1/lib

nca@nca:~/qgroundcontrol/build$ cmake -DCMAKE_BUILD_TYPE=Debug ..
nca@nca:~/qgroundcontrol/build$ make -j4  or   make -j$(nproc)
.
.
.
[ 57%] Building CXX object src/Utilities/CMakeFiles/Utilities.dir/QGCCachedFileDownload.cc.o
/home/nca/qgroundcontrol/src/Utilities/KMLHelper.cc: In static member function ‘static QDomDocument KMLHelper::_loadFile(const QString&, QString&)’:
/home/nca/qgroundcontrol/src/Utilities/KMLHelper.cc:33:57: warning: ‘bool QDomDocument::setContent(QIODevice*, QString*, int*, int*)’ is deprecated: Use the overload returning ParseResult instead. [-Wdeprecated-declarations]
   33 |     if (!doc.setContent(&file, &errorMessage, &errorLine)) {
      |                                                         ^
In file included from /usr/local/qt6.8.1/include/QtXml/QDomDocument:1,
                 from /home/nca/qgroundcontrol/src/Utilities/KMLHelper.h:13,
                 from /home/nca/qgroundcontrol/src/Utilities/KMLHelper.cc:10:
/usr/local/qt6.8.1/include/QtXml/qdom.h:330:10: note: declared here
  330 |     bool setContent(QIODevice *dev, QString *errorMsg, int *errorLine = nullptr, int *errorColumn = nullptr);
      |          ^~~~~~~~~~
[ 57%] Building CXX object src/Utilities/CMakeFiles/Utilities.dir/QGCFileDownload.cc.o
[ 57%] Building CXX object src/Utilities/CMakeFiles/Utilities.dir/QGCLoggingCategory.cc.o
[ 57%] Building CXX object src/Utilities/CMakeFiles/Utilities.dir/QGCTemporaryFile.cc.o
[ 57%] Building CXX object src/Utilities/CMakeFiles/Utilities.dir/ShapeFileHelper.cc.o
/home/nca/qgroundcontrol/src/Utilities/QGCFileDownload.cc: In static member function ‘static void QGCFileDownload::setIgnoreSSLErrorsIfNeeded(QNetworkReply&)’:
/home/nca/qgroundcontrol/src/Utilities/QGCFileDownload.cc:38:40: error: ‘QSslSocket’ has not been declared
   38 |     const bool sslLibraryBuildIs1x = ((QSslSocket::sslLibraryBuildVersionNumber() & 0xf0000000) == 0x10000000);
      |                                        ^~~~~~~~~~
/home/nca/qgroundcontrol/src/Utilities/QGCFileDownload.cc:39:35: error: ‘QSslSocket’ has not been declared
   39 |     const bool sslLibraryIs3x = ((QSslSocket::sslLibraryVersionNumber() & 0xf0000000) == 0x30000000);
      |                                   ^~~~~~~~~~
/home/nca/qgroundcontrol/src/Utilities/QGCFileDownload.cc:43:56: error: incomplete type ‘QSslError’ used in nested name specifier
   43 |         errorsThatCanBeIgnored << QSslError(QSslError::NoPeerCertificate);
      |                                                        ^~~~~~~~~~~~~~~~~
/home/nca/qgroundcontrol/src/Utilities/QGCFileDownload.cc:43:73: error: invalid use of incomplete type ‘class QSslError’
   43 |         errorsThatCanBeIgnored << QSslError(QSslError::NoPeerCertificate);
      |                                                                         ^
In file included from /usr/local/qt6.8.1/include/QtNetwork/QNetworkAccessManager:1,
                 from /usr/local/qt6.8.1/include/QtNetwork/qnetworkreply.h:13,
                 from /usr/local/qt6.8.1/include/QtNetwork/QNetworkReply:1,
                 from /home/nca/qgroundcontrol/src/Utilities/QGCFileDownload.h:14,
                 from /home/nca/qgroundcontrol/src/Utilities/QGCFileDownload.cc:10:
/usr/local/qt6.8.1/include/QtNetwork/qnetworkaccessmanager.h:29:7: note: forward declaration of ‘class QSslError’
   29 | class QSslError;
      |       ^~~~~~~~~
/home/nca/qgroundcontrol/src/Utilities/QGCFileDownload.cc:44:60: error: no matching function for call to ‘QNetworkReply::ignoreSslErrors(QList<QSslError>&)’
   44 |         networkReply.ignoreSslErrors(errorsThatCanBeIgnored);
      |                                                            ^
In file included from /usr/local/qt6.8.1/include/QtNetwork/QNetworkReply:1,
                 from /home/nca/qgroundcontrol/src/Utilities/QGCFileDownload.h:14,
                 from /home/nca/qgroundcontrol/src/Utilities/QGCFileDownload.cc:10:
/usr/local/qt6.8.1/include/QtNetwork/qnetworkreply.h:125:18: note: candidate: ‘virtual void QNetworkReply::ignoreSslErrors()’
  125 |     virtual void ignoreSslErrors();
      |                  ^~~~~~~~~~~~~~~
/usr/local/qt6.8.1/include/QtNetwork/qnetworkreply.h:125:18: note:   candidate expects 0 arguments, 1 provided
In file included from /usr/local/qt6.8.1/include/QtCore/qglobal.h:13,
                 from /usr/local/qt6.8.1/include/QtCore/qloggingcategory.h:7,
                 from /usr/local/qt6.8.1/include/QtCore/QLoggingCategory:1,
                 from /home/nca/qgroundcontrol/src/Utilities/QGCFileDownload.h:12,
                 from /home/nca/qgroundcontrol/src/Utilities/QGCFileDownload.cc:10:
/usr/include/c++/9/type_traits: In instantiation of ‘struct std::is_trivial<QSslError>’:
/usr/include/c++/9/type_traits:2889:25:   required from ‘constexpr const bool std::is_trivial_v<QSslError>’
/usr/local/qt6.8.1/include/QtCore/qtypeinfo.h:56:27:   required from ‘class QTypeInfo<QSslError>’
/usr/local/qt6.8.1/include/QtCore/qarraydataops.h:859:33:   required by substitution of ‘template<class T> struct QtPrivate::QArrayOpsSelector<T, typename std::enable_if<(QTypeInfo<T>::isComplex && QTypeInfo<T>::isRelocatable)>::type> [with T = QSslError]’
/usr/local/qt6.8.1/include/QtCore/qarraydataops.h:866:8:   required from ‘struct QtPrivate::QCommonArrayOps<QSslError>’
/usr/local/qt6.8.1/include/QtCore/qarraydataops.h:950:8:   required from ‘struct QArrayDataOps<QSslError>’
[... 139 lines cut ...]
                 from /usr/local/qt6.8.1/include/QtCore/qloggingcategory.h:8,
                 from /usr/local/qt6.8.1/include/QtCore/QLoggingCategory:1,
                 from /home/nca/qgroundcontrol/src/Utilities/QGCFileDownload.h:12,
                 from /home/nca/qgroundcontrol/src/Utilities/QGCFileDownload.cc:10:
/usr/local/qt6.8.1/include/QtCore/qarraydataops.h: In instantiation of ‘struct QtPrivate::QGenericArrayOps<QSslError>’:
/usr/local/qt6.8.1/include/QtCore/qarraydataops.h:646:8:   required from ‘struct QtPrivate::QMovableArrayOps<QSslError>’
/usr/local/qt6.8.1/include/QtCore/qarraydataops.h:866:8:   required from ‘struct QtPrivate::QCommonArrayOps<QSslError>’
/usr/local/qt6.8.1/include/QtCore/qarraydataops.h:950:8:   required from ‘struct QArrayDataOps<QSslError>’
/usr/local/qt6.8.1/include/QtCore/qarraydatapointer.h:109:22:   required from ‘QArrayDataPointer<T>::~QArrayDataPointer() [with T = QSslError]’
/usr/local/qt6.8.1/include/QtCore/qlist.h:277:5:   required from here
/usr/local/qt6.8.1/include/QtCore/qarraydataops.h:287:25: error: static assertion failed: Types with throwing destructors are not supported in Qt containers.
  287 |     static_assert (std::is_nothrow_destructible_v<T>, "Types with throwing destructors are not supported in Qt containers.");
      |                    ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/qt6.8.1/include/QtCore/qarraydataops.h: In instantiation of ‘struct QtPrivate::QMovableArrayOps<QSslError>’:
/usr/local/qt6.8.1/include/QtCore/qarraydataops.h:866:8:   required from ‘struct QtPrivate::QCommonArrayOps<QSslError>’
/usr/local/qt6.8.1/include/QtCore/qarraydataops.h:950:8:   required from ‘struct QArrayDataOps<QSslError>’
/usr/local/qt6.8.1/include/QtCore/qarraydatapointer.h:109:22:   required from ‘QArrayDataPointer<T>::~QArrayDataPointer() [with T = QSslError]’
/usr/local/qt6.8.1/include/QtCore/qlist.h:277:5:   required from here
/usr/local/qt6.8.1/include/QtCore/qarraydataops.h:649:25: error: static assertion failed: Types with throwing destructors are not supported in Qt containers.
  649 |     static_assert (std::is_nothrow_destructible_v<T>, "Types with throwing destructors are not supported in Qt containers.");
      |                    ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [src/Utilities/CMakeFiles/Utilities.dir/build.make:180: src/Utilities/CMakeFiles/Utilities.dir/QGCFileDownload.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:7316: src/Utilities/CMakeFiles/Utilities.dir/all] Error 2
make: *** [Makefile:166: all] Error 2

HTRamsey, 2025-01-22:
If you are trying to build the daily, we had to bump the minimum to Ubuntu 22.04: https://doc.qt.io/qt-6/linux.html#supported-configurations

hesamira, 2025-01-23:
Thank you for your guidance.👍

closed 2025-01-23 by HTRamsey