mavlink/qgroundcontrol #12372: Daily build fails with "exiv2.dll was not found" error

rmackay9, 2025-01-30:
On my Windows11 machine, I uninstalled the stable QGC (which works fine) and re-installed the Daily build ([linked from this wiki page](https://docs.qgroundcontrol.com/master/en/qgc-user-guide/releases/daily_builds.html)) but sadly QGC fails to start properly.  It complains that, "exiv2.dll was not found"

[image]

hyeonseung89, 2025-01-30:
I'm experiencing the same problem. I reverted the commit ' fix window debug build '(c71b0d3) and it works. Can you confirm it?

closed 2025-01-31 by DonLakeFlyer

rmackay9, 2025-02-03:
Hi there, thanks very much for fixing the above issue.  There's sadly another issue that is stopping the Daily build from running on Windows which I've added as a new issue here https://github.com/mavlink/qgroundcontrol/issues/12389

Happy to test any fixes that might help, sorry I'm not really too qualified to dig into it myself

HTRamsey, 2025-02-03:
@hyeonseung89 can you verify?

PR #12376 (2025-01-31): Build Exiv2 as static lib
Fix #12372