mavlink/qgroundcontrol #12378: Review exiv2 Windows hack

DonLakeFlyer, 2025-01-31:
Take a look at #12376 to see if it should be modified to something else

s-lisovenko, 2025-02-01:
I may be wrong, but I think we can't use static linking for this library and distribute QGC under LGPL and Apache licenses, since exiv2 is only distributed under GPL 2 license.

HTRamsey, 2025-02-01:
QGC can't be used under LGPL right now anyways

s-lisovenko, 2025-02-01:
Ah, you are right, qgc use gplv3, but for mobile apache 2.0 right? And statically linking gplv2 and apache 2.0 incompatibility too if i understand correctly.

DonLakeFlyer, 2025-02-02:
So is there a problem here or not? If the licensing is screwed up what needs fixing. The deployment targets are:
* Installers downloaded from internet
* Android, Mac, Windows store

Each of which may have differing license targets. Ages ago someone said we couldn't deploy to Mac store without using a Qt commercial license for example. No idea if that's a true statement or not.

s-lisovenko, 2025-02-02:
> So is there a problem here or not? If the licensing is screwed up what needs fixing. The deployment targets are:
>
> * Installers downloaded from internet
> * Android, Mac, Windows store
>
> Each of which [...]

Yes, AppleStore might be the biggest problem, need to check that, but as far as I know (may be wrong) in the past it was very difficult to get a gplv2 app there.

HTRamsey, 2025-02-02:
You'd have to remove several of the Qt libraries we use as well. This goes back to #11600 where I tried to remove non-critical code like the Mavlink Message charting to remove some GPL dependencies, but that adds complexity to some extent. This is a requirement for a lot of vendors, but also not really the responsibility of QGC maintainers to provide in my opinion. It is nice to make it easier for them to do though. However, this removes some incentive to contribute back to project too

s-lisovenko, 2025-02-02:
> You'd have to remove several of the Qt libraries we use as well. This goes back to [#11600](https://github.com/mavlink/qgroundcontrol/pull/11600) where I tried to remove non-critical code like the [...]

I think the problem is that even using a qt commercial license exiv2 will not give you the option of any other license than gpl2 when statically linking.

HTRamsey, 2025-02-02:
Yeah, we could look at shared linking but that has some consequences. There are other exif libs but from what I saw they generally only allow reading and not writing. And manually parsing/writing exif is error prone and not exactly simple to expand on.

closed 2025-03-03 by HTRamsey