mavlink/qgroundcontrol #12389: Daily build starts process but main window never appears

rmackay9, 2025-02-03:
In a follow up to Issue https://github.com/mavlink/qgroundcontrol/issues/12372 I've retested the [daily build](https://docs.qgroundcontrol.com/Stable_V4.3/en/qgc-user-guide/releases/daily_builds.html) for windows and while it doesn't fail in the same way as before, it is still sadly unusable on Windows.

The issue now is that the QGC process starts, I can see it in the Tasks Manager list but the main window never appears.  I've retested with Stable and QGC does start correctly so I'm certain it is an issue with the Daily build.

DonLakeFlyer, 2025-02-03:
That is probably fixed by #12384

DonLakeFlyer, 2025-02-03:
There might still be an issue here. On Mac the app is missing it's icons. Not sure if Windows is the same.

DonLakeFlyer, 2025-02-03:
This hopefully last set of fixes in #12387 should fix missing icons. @rmackay9 Can you give it a try?

rmackay9, 2025-02-03:
@DonLakeFlyer,

Thanks very much for the help.  I've just [downloaded the daily build](https://docs.qgroundcontrol.com/Stable_V4.3/en/qgc-user-guide/releases/daily_builds.html) and tried again and sadly no luck.  QGC still starts up in the background but the main window doesn't appear

DonLakeFlyer, 2025-02-04:
Hmm, I can't seem to reproduce. Two things:
* There is a delay between S3 storage and cloudfront copies which the web site uses. Maybe it hand't moved up yet. Let's remove the web site from the equation and pull the installer from github actions last master build instead.
* If it still doesn't work send me a dir listing of QGC in program files. There must be some dll missing in there.

DonLakeFlyer, 2025-02-04:
@HTRamsey Do you have a windows machine? If so, cam you try the windows build?

HTRamsey, 2025-02-04:
Yup, I'll try. Just to clarify, is this specifically Windows 10/11 or possibly both?

DonLakeFlyer, 2025-02-04:
Not sure what @rmackay9 has. I've tried on 11.

HTRamsey, 2025-02-04:
@rmackay9 did you also try the GPU safe mode exe? They manually set AA_UseDesktopOpenGL/AA_UseSoftwareOpenGL. The other thing to try would be adding to the opengl/buglist.json.

HTRamsey, 2025-02-04:
Just tried and it works on Windows 10 using the standard exe & GPU compatibility mode but not safe mode. Also tried running on both the intel GPU and an nvidia quadro t2000 and it worked on both. I think I've had issues running it in a VM in the past though. That may be related to `QQuickWindow::setGraphicsApi(QSGRendererInterface::OpenGL);` but I'm not sure. Need to try it with GStreamer disabled sometime.

HTRamsey, 2025-02-04:
@DonLakeFlyer I think we need to manually link against OpenGL with GStreamer. Could you try adding
user32.lib & OpenGL::GL to the gstreamer linking? Need to add OpenGL to FindPackage too

rmackay9, 2025-02-04:
Hi there, it's opening for me now (Windows 11).  I don't think I did anything different from previous tests and I think I was pretty careful.  The only thing that gets me sometimes is that immediately after the first install it takes a while to startup (20 seconds), but I'm sure I must have waited that long in previous tests as well.

In any case, it's all fine now and sorry if I chewed up people's time on this.  Thanks!

closed 2025-02-04 by rmackay9

DonLakeFlyer, 2025-02-05:
> Hi there, it's opening for me now (Windows 11).

The same build or did you install a new one?

rmackay9, 2025-02-06:
Hi @DonLakeFlyer, I uninstalled what was on my PC and then downloaded a new .msi from [the page ](https://docs.qgroundcontrol.com/Stable_V4.3/en/qgc-user-guide/releases/daily_builds.html), and installed it.

Txs again

DonLakeFlyer, 2025-02-06:
Ok, thanks Randy. I think it was just a delay on S3->Cloudfront then and I really did fix the damn thing!