mavlink/qgroundcontrol #12390: QGC Crash

G-IMP, 2025-02-03:
QGC Crashes when flashing PX4 Firmware, or when connecting QGC
Note: This doesn't happen on my old laptop with a 4.3.0 QGC using the same Pixhawk

It should displays this error but QGC crashes before it pops up
[image]

System Information
- Windows 11
- Latest QGC & PX4 Stable release
- CUAV v6x

HTRamsey, 2025-02-03:
So you're running 4.4.X on Windows 11 correct?

DonLakeFlyer, 2025-02-03:
The missing parameters stuff yes was taken care of in daily. PX4 firmware moed some params around. I didn't think it was causing crashes though.

G-IMP, 2025-02-04:
> The missing parameters stuff yes was taken care of in daily. PX4 firmware moed some params around. I didn't think it was causing crashes though.

I see, well i still got the crashes with other popup windows after flashing. Fixed it with making sure QGC was running on NVIDIA

closed 2025-02-04 by G-IMP

DonLakeFlyer, 2025-02-05:
Is it a laptop with Intel graphics on it?

G-IMP, 2025-02-05:
> Is it a laptop with Intel graphics on it?

AMD Ryzen