mavlink/qgroundcontrol #12396: QGC 5.0 release

gillamkid, 2025-02-05:
It's been a while since QGC master was turned into a release (last time was more than a year ago, Nov 2023 v4.3.0)

For my company, we have a fork of QGC based off the latest 4.x releases. We are anxious for a new 5.0 release so we can have a stabilized version of updates in QGC master we've been drueling over 🤤. Our roadmap relies on getting some of these features into our fork, such as the updates to some library versions.

Can a plan be made to prepare to release QGC 5.0 soon? I'd love to get a release in a couple months. What would have to happen to begin the process of release?

@DonLakeFlyer 
@HTRamsey 
@mrpollo 
@Davidsastresas

HTRamsey, 2025-02-05:
I believe we are trying to release in sync with px4, so should be a couple months

gillamkid, 2025-02-05:
@HTRamsey Are you referring to the upcoming PX4 1.16 release? I've been going to the PX4 weekly sync meetings regularly for a few months now. I know there are some dedicated engineers flight testing 1.16 seeing if it is ready for release. I could double check, but I'd be very surprised if they were using QGC daily to test. I've found in the weekly PX4 call there isn't anyone working closely with QGC upstream.

At the next PX4 weekly sync, should I tell them that QGC wants to release 5.0 with 1.16 and ask those flight testing 1.16 to start using QGC daily? Should we make a QGC v5.0.0-Beta1 for them to start using?

DonLakeFlyer, 2025-02-05:
Yes, they should be using daily builds to flight test. No need to make a special beta thing.

DonLakeFlyer, 2025-05-05:
Goal is the cut first RC of 5.0 next week.

closed 2025-05-15 by DonLakeFlyer