mavlink/qgroundcontrol #12405: Feedback on new Toolbar Dropdown expanded pages

DonLakeFlyer, 2025-02-06:
A significant usability change in the 5.0 version QGC are the new toolbar dropdowns. They now have a section which can be expanded which can contain both application and vehicle settings associated with the indicator. The goal of this is to provide to user with access to settings which are of use to be changed from one flight to the next. As opposed to things like the full "Vehicle Setup" pages which mostly contains things which are of use when designing/creating a new vehicle.

Here is an example of the PX4 Battery Indicator with the pane expanded:

[image]

DonLakeFlyer, 2025-02-06:
So the feedback I am looking for is what is missing from these expanded pages. As well as what is there that shouldn't. And most importantly why?

DroneWuKong, 2025-02-06:
Hello, 

Taking a quick look at it, It would be nice to be able to display average single cell voltage as regardless of battery pack total voltage individual cell voltage stays the same. In addition to being able to display current draw.

With the ability to actively monitor current draw would allow a operator to have an understanding of how hard the motors/batteries are working in regards to load. Under varying weather conditions amp draw can be used to estimate if flight times are shortened or extended at a quick glance, in addition to a better ability to fine tune cruising speed in relation power draw.

AndKe, 2025-02-12:
I like it, very nice and useful.
as @DroneWuKong mentions , cell voltage is a kind of "universal language" for those who switch between 4s,5s,6s,12s UAV's 
Also seeing ThO plot (throttle-output)  is a good visualisation of the actual power.
in planes ThO near 100%  in cruise can tell you that the target groundspeed is too high for certain winds, or that the battery voltage is soon to low to maintain altitude, an elevated ThO can indicate icing too.  In general, plotting/seeing ThO is a useful tool in terms of power.

DonLakeFlyer, 2025-02-12:
> cell voltage is a kind of "universal language" for those who switch between

Yeah, that is very doable. I'll add it.

Where does throttle output come from mavlink. It's not available to be added to the instrument panel?

DonLakeFlyer, 2025-02-12:
Also keep in mind that all of the toolbar dropdowns have new expanded pages. Not just the battery example I posted above.

AndKe, 2025-02-12:
Sorry the ThO is of course from datalogs.
Throttle demand in telemetry is logged in VRR_HUD.throttle  for copters and planes  (except plane in VTOL mode.)

AndKe, 2025-02-12:
As for the other expandable panes:  I love them - especially the failsafe options. 
Also: on Linux I see font size/alignment issue:  (this issue is nowhere else in the app as I can see)

[image]

Please consider making the "expansion arrow" more prominent - this deserves to be more accessible/easier to discover.

[image]

DonLakeFlyer, 2025-02-12:
> Please consider making the "expansion arrow" more prominent - this deserves to be more accessible/easier to discover.

Yeah, been meaning to do this forever. Royal PITA to hit it on my Herelink!

DonLakeFlyer, 2025-02-12:
On the cell voltage thing this is easily doable on PX4 since there is a standard vehicle param for cell count which is configured in Power setup. But on ArduPilot I can't seem to find the same sort of param to use.

AndKe, 2025-02-13:
Yes, Ardupilot does not show cell count, and unless it senses the cell voltages, there is no proper way to know.
I had another telemetry-based project that always got the detection right it worked like this:
on startup, if voltage is:
10 -12.7v = 3S
12.7 - 17v =4S
17--21.8v = 5s
21.7-25.5 =6s
38.4-51.0=12s
 
basically assuming that about 3.2v ...4.35v  is the working rage of each cell.   I did not care much for LIHV, sp the numbers may need to be adjusted a little for that.

AndKe, 2025-02-13:
BTW: Ninja builds work fine, but are slow, using only a few cores for most of the job.

[image]

DonLakeFlyer, 2025-02-13:
> BTW: Ninja builds work fine, but are slow, using only a few cores for most of the job.

@HTRamsey Normal or not?

DonLakeFlyer, 2025-02-14:
I notice the same thing on my Mac. The CPUs are pegged only a minority of the time.

closed 2025-05-03 by HTRamsey