mavlink/qgroundcontrol #12417: Missing "Goto Location" menu in Fly View

JustThinkInC, 2025-02-10:
I am unable to get the menu for "Go to location" as per the [docs](https://docs.qgroundcontrol.com/Stable_V4.3/en/qgc-user-guide/fly_view/fly_view.html#goto) on the Pixhawk 6C running ArduPilot 4.5.7

## Expected Behavior
I should see a "Go to" menu option when click on the map after a takeoff.

## Current Behavior
I only see the options to set home and estimator origin.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Takeoff in (or switch to) GUIDED mode
2. Click on map near vehicle location
3. Cannot see go to menu

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10 & 11
- QGC Version: 4.4.3
- QGC build: stable & self-built from source
- Flight Controller: [e.g., Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.]
- Autopilot (with version): Pixhawk 6C with Ardupilot 4.5.7

DonLakeFlyer, 2025-02-11:
Using a daily build do the following:
* Boot QGC with the vehicle turned off, so no connection
* Turn on Console Logging for `GuidedActionsControllerLog`. (GuidedActionsControllerLog)
* Turn on the vehicle so it will connect
* Takeoff
* Click on map to attempt GoTo location
* Go back to Console, save the log and post here

JustThinkInC, 2025-02-12:
Thank you, installed another daily build today and ran. I've attached the log here, kept it hovering for a few mins and clicking on the map but nothing.

[gotolog.txt](https://github.com/user-attachments/files/18765274/gotolog.txt)

JustThinkInC, 2025-02-13:
OK I will send the telemetry log file privately (don't really want to give my location to everyone).

DonLakeFlyer, 2025-02-16:
Since ArduPilot doesn't support `MAVLINK_MSG_ID_EXTENDED_SYS_STATE` QGC determines whether the vehicle is flying or not from `HEARTBEAT.system_status`. It is looking for a `MAV_STATE` of `MAV_STATE_ACTIVE` which is 4. In your telemetry log the vehicle is sending `MAV_STATE_CRITICAL` which is 5. I think it might have an active RC failsafe going on. The definition of 5 is "System is in a non-normal flight mode (failsafe). It can however still navigate.". QGC doesn't interpret that case as "flying". But from the mavlink spec I guess it sort of indicates "flying" maybe. I"ll add support for `MAV_STATE_CRITICAL` which I think is ok.

Do you understand why you were in a failsafe mode?

closed 2025-02-16 by DonLakeFlyer

JustThinkInC, 2025-02-17:
Gotcha! Ah I wanted to perform the flight without an RC transmitter and just the radio telem so that'll explain the failsafe.

PR #12465 (2025-02-16): Fix ArduPilot "flying" determination
Fix #12417