mavlink/qgroundcontrol #12426: Android: Can't enter a server in a UDP Comm link

timtuxworth, 2025-02-12:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/support/support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
I should be able to enter a server for a UDP comm link

## Current Behavior
Whatever I type in the server field is ignored

## Steps to Reproduce:
Create  a new or enter a new COmm link
Select UDP if it's new
click in the Server field
Type in the server name e.g. 192.168.144.20:19956
Click "Done" or click the check mark on the keyboard.
The entire comm link dialog closes, the server name entered is not saved.

## System Information
SIYI Uni RC 7 running Android 13
QGC Daily Stable or Master from "Android-Linux

Here's a video

https://github.com/user-attachments/assets/37d346dd-4fc8-46c6-aa97-823f69904da3

DonLakeFlyer, 2025-02-12:
Hmm, strange. The close on Done repros on my Herelink as well. I'd guess the massive virtual keyboard is maybe some sort of Android settings. Not sure. I'm, hoping to get a Uni RC 7 soon as well so I can try on that.

closed 2025-02-12 by DonLakeFlyer

timtuxworth, 2025-02-13:
> Hmm, strange. The close on Done repros on my Herelink as well. I'd guess the massive virtual keyboard is maybe some sort of Android settings. Not sure. I'm, hoping to get a Uni RC 7 soon as well so I can try on that.

Yes - I don't have the best eyesight, I have increased the size from standard so that I can see clearly, but it's all standard Android settings, so the app should work no?

DonLakeFlyer, 2025-02-13:
This should be fixed now. For the large virtual keyboard I don't see how QGC has any control over that.

PR #12428 (2025-02-12): Remove enter key to close dialogs
This would lead to unexpected dialog close both on desktop and mobile. But much worse on mobile with virtual keyboard. Also I think with the new Qt version something changed with the virtual keyboard simulating an enter key press.

Fix #12426