mavlink/qgroundcontrol #12429: Missing showError param causes error

aXynt, 2025-02-12:
Missing showError parameter for sendMavCommand function causes error in function sendMavCommandWithLambdaFallback
https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L2471

HTRamsey, 2025-02-12:
It would be very helpful if you could make a PR for this, otherwise I'll get to it soon

HTRamsey, 2025-02-12:
@DonLakeFlyer should showError be true or false?

DonLakeFlyer, 2025-02-12:
This should work like the other sendMavCommand methods which don't have a handler to deal with command results. Specifically the signature should include showError since it's up to the caller as to whether they want that or not:
```
    void sendMavCommandWithLambdaFallback(
        std::function<void()> lambda,
        int compId, MAV_CMD command,
        bool showError,
        ...
```

Just like all of these do:
```
    void sendMavCommand(int compId, MAV_CMD command, bool showError, float param1 = 0.0f, float param2 = 0.0f, float param3 = 0.0f, float param4 = 0.0f, float param5 = 0.0f, float param6 = 0.0f, float param7 = 0.0f);
    void sendMavCommandDelayed(int compId, MAV_CMD command, bool showError, int milliseconds, float param1 = 0.0f, float param2 = 0.0f, float param3 = 0.0f, float param4 = 0.0f, float param5 = 0.0f, float param6 = 0.0f, float param7 = 0.0f);
    void sendMavCommandInt(int compId, MAV_CMD command, MAV_FRAME frame, bool showError, float param1, float param2, float param3, float param4, double param5, double param6, float param7);
```

And then in the specific use case of setCurrentMissionSequence it should be called with showError = true.

PR #12437 (2025-02-13): Add showError support to sendMavCommandWithLambdaFallback
Fix #12429
closed 2025-02-13 by DonLakeFlyer