mavlink/qgroundcontrol #12439: PX4 airframe selection submenu is broken

dakejahl, 2025-02-12:
The drop down menu doesn't work on the PX4 airframe selection page. Depending on which airframe you select you get a variable number of these warnings
```
qrc:/qml/QGroundControl/Controls/QGCComboBox.qml:80:9: Unable to assign [undefined] to QString
qrc:/qml/QGroundControl/Controls/QGCComboBox.qml:80:9: Unable to assign [undefined] to QString
```

[image]

dakejahl, 2025-02-12:
The next question is what happened to the images for the airframes?

DonLakeFlyer, 2025-02-12:
I forgot to mention that. The airframe metadata xml I get from PX4 no longer has image tags defined in it. So that's a PX4 thing.

DonLakeFlyer, 2025-02-12:
@dakejahl Can you check the meta data generation on the firmware side for me?

hamishwillee, 2025-02-12:
@DonLakeFlyer @dakejahl FWIW The files that generate this haven't changed for at least 3 years - such as https://github.com/PX4/PX4-Autopilot/blob/main/Tools/px_process_airframes.py and https://github.com/PX4/PX4-Autopilot/blob/main/Tools/px4airframes/xmlout.py

I just ran this on a recent PX4 `main` and I can see image tags in the XML like `<airframe_group name="Autogyro" image="Autogyro">`

DonLakeFlyer, 2025-02-13:
Maybe something wrong with metadata cache then. The file I have doesn't have them.

DonLakeFlyer, 2025-02-13:
Yeah, sorry about that @hamishwillee @dakejahl. I was looking at the wrong thing. Resource names changed for airframes on QGC side.

closed 2025-02-13 by DonLakeFlyer

hamishwillee, 2025-02-20:
No worries. Good to see so much activity in QGC land.

PR #12440 (2025-02-12): ComboBox: Fix handling of text role in model data
Fix #12439

@dakejahl Can you verify this as well once a build pops out?
PR #12444 (2025-02-13): Fix airframe image resource names
Fix #12439

[image]