mavlink/qgroundcontrol #12449: Android icons bug

s-lisovenko, 2025-02-13:
Android icons rendering problem

## Steps to Reproduce:
1. Download artifact and install
2. Set up App (during first lauch everything is ok
3. Close app
4. Run app again

## System Information
When posting bug reports, include the following information
- Android 12 Samsung Galaxy
- QGC Version: latest
- QGC build: daily
- Teststed artifact(linux release for android):
https://github.com/mavlink/qgroundcontrol/actions/runs/13300635398/artifacts/2583443183

## Detailed Description

[image]

HTRamsey, 2025-02-13:
When was the last time you tested it? That's super weird

s-lisovenko, 2025-02-13:
[image]

s-lisovenko, 2025-02-13:
To be honest, I don't know. I think something changed with qt 6.8.2. I tried migrating my app and found that strange behavior. 6.8.0 works as expected. Not sure about 6.8.1 as we don’t use it in production.

HTRamsey, 2025-02-13:
Nevermind I just found the issue

HTRamsey, 2025-02-13:
Or maybe I didn't, not sure if I can reproduce. You can try changing this and building on 6.8.0 if you want:

[image]

HTRamsey, 2025-02-13:
some this (https://github.com/mavlink/qgroundcontrol/commit/429b87c43919f8ebd621f0c93cee4734a9a41dde) drastically reduced the Android size. So I'm making a PR to revert that and may get you to try again

s-lisovenko, 2025-02-13:
I'm almost sure that problem not related to idk, I have the same behavior when I compile project in Qt Creator with right android tools

s-lisovenko, 2025-02-13:
I'll try to contact Qt support, since I have a commercial license, maybe something has changed since 6.8.0

HTRamsey, 2025-02-13:
okay, I tried building locally and using a CI built APK without issue. We'll get @DonLakeFlyer to try on his herelink later too

s-lisovenko, 2025-02-13:
The same build works well on Android14

HTRamsey, 2025-02-13:
ohhh, wait which android version were you using?

s-lisovenko, 2025-02-13:
But 6.8.0 version works well on both. Don’t know why.

HTRamsey, 2025-02-13:
weird, yeah I would probably have to chalk that up to a Qt issue. There are some QSG related env values that you could try to get some debug output from, but I'm not sure of a solution. For now you could just force 6.8.0 using those CMake variables I posted. There's not really any implementation differences between qt minor versions

s-lisovenko, 2025-02-13:
Of course. But I really want to know what is the problem. I will create issue for qt support, maybe this is really new bug. Or maybe it related only ro my device. Will be good if someone can test on another Android

DonLakeFlyer, 2025-02-13:
I don't have anything running 12. My Galaxy running 14 looks fine.

s-lisovenko, 2025-03-03:
This is most likely a QT bug. So I close this issue.

closed 2025-03-03 by s-lisovenko