mavlink/qgroundcontrol #12462: QGC does not connect to default port 14550

ytl001, 2025-02-14:
## Expected Behavior
I am running gazebo classic simulator in a docker container (image: jonasvautherin/px4-gazebo-headless).  QGC running on the same host as docker should automatically connect to default port 14550.

## Current Behavior
QGC does not connect.  Console shows (with qgc.comms.linkmanager and qgc.comms.linkmanager.verbose on):

[image]

I tried turning autoconnect all on, all off except udp, all off but with a manual link for port 14550 (as below).  All scenarios produce the same results.

[image]

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.  sudo /usr/local/opt/docker-mac-net-connect/bin/docker-mac-net-connect
2.  docker run --rm -it jonasvautherin/px4-gazebo-headless 172.17.0.1 172.17.0.1
3.  click on QGC icon to start QGC

## System Information
When posting bug reports, include the following information
- Operating System: macOS Ventura 13.7.2
- QGC Version:  Daily version a8ab437 64 bit, downloaded Feb 13, 2025
- QGC build: 
- Flight Controller: PX4_SITL
- Autopilot (with version): px4 release 1.15.3 (17761279)

HTRamsey, 2025-02-15:
I think this is docker related but I don't recall the solution right now. Did you try with the last stable build?

ytl001, 2025-02-15:
Yes, I tried using the last stable build v4.4.3.  Results are the same, except that with Linkmanagerverbose on, console seems to show that QGC is listening to a bluetooth port only:

[image]

I am running docker-mac-net-connect to make sure that the host can reach the container.  Indeed I can ping the container from the host and when I set up a comm link to point to the container port assigned in px4, px4 reports a valid ground station heart beat (but QGC remains unconnected).

ytl001, 2025-02-18:
Thank you DonLakeFlyer and HTRamsey!

I tried setting MAV_#_BROADCAST to 1 in PX4 (inside docker) but mavlink status still showed "Broadcast enabled: No" for some reason.  Another person had the same issue: https://discuss.px4.io/t/cannot-enable-mav-i-broadcast/22408.  He used the same solution as DonLakeFlyer.  I might have to learn how to make my own docker image.

DonLakeFlyer, 2025-02-18:
Closing since I'm pretty sure this is a udp broadcast problem

closed 2025-02-18 by DonLakeFlyer

ytl001, 2025-03-04:
I made the code change in ROMFS/px4fmu_common/init.d-posix/px4-rc.mavlink as indicated in https://github.com/mavlink/qgroundcontrol/issues/12462#issuecomment-2661032768.  Now broadcast is enabled in px4 but QCS is still not connecting:  

[image]

I checked that the host port 14550 was indeed open and listening:

[image]

I could ping the host inside the container running px4 and could ping the container on the host.  In the container, port 18570 seemed open and connected to remote port 14550 (from the mavlink status info above).  Both docker and QCS are on the list of "allowing incoming connections" in the firewall setting.  Yet something seems blocking the connection.  What did I miss?