mavlink/qgroundcontrol #12471: ODOMETRY unsupported estimator type 8  issue

tkddus21, 2025-02-17:
what's problem with this...?

tkddus21, 2025-02-17:
[Screenshot from 2025-02-17 16-13-17.zip](https://github.com/user-attachments/files/18821386/Screenshot.from.2025-02-17.16-13-17.zip)

DonLakeFlyer, 2025-02-17:
As is says in the message. This is a Vehicle Error. You are going to have to ask the firmware folks.

closed 2025-02-17 by DonLakeFlyer

TSC21, 2025-02-17:
This is most likely a loopback issue, i.e. the data coming out of the FMU (PX4 in this case) is probably being fedback into the FMU again. That is why the frame of reference is not supported.

ceydanuroncel137, 2025-03-27:
I have the same problem. I use orange cube plus. I use px4 version 1.14.2, gazebo 11 and qground control v4.4.3 version. Actually making HITL connection. But even if move the drone joystick, it never moves. I checked all my connections and sensors and now ı'm getting an odometry sensor.