mavlink/qgroundcontrol #12477: I am running QGC on an Android 14 pad and have a radio connected to the serial port, but cannot enumerate the USB serial devices

xzxzhao2013, 2025-02-19:
console log：
W/qgc.android.androidserial: availableDevicesInfo returned null
W/default : java.lang.SecurityException: User has not given 10152/org.mavlink.qgroundcontrol permission to access device /dev/bus/usb/002/006
W/default : 	at android.os.Parcel.createExceptionOrNull(Parcel.java:3057)
W/default : 	at android.os.Parcel.createException(Parcel.java:3041)
W/default : 	at android.os.Parcel.readException(Parcel.java:3024)
W/default : 	at android.os.Parcel.readException(Parcel.java:2966)
W/default : 	at android.hardware.usb.IUsbSerialReader$Stub$Proxy.getSerial(IUsbSerialReader.java:126)
W/default : 	at android.hardware.usb.UsbDevice.getSerialNumber(UsbDevice.java:157)
W/default : 	at org.mavlink.qgroundcontrol.QGCUsbSerialManager.formatDeviceInfo(QGCUsbSerialManager.java:441)
W/default : 	at org.mavlink.qgroundcontrol.QGCUsbSerialManager.availableDevicesInfo(QGCUsbSerialManager.java:423)
W/default : 	at org.qtproject.qt.android.QtNative.startQtApplication(Native Method)
W/default : 	at org.qtproject.qt.android.QtNative$$ExternalSyntheticLambda4.run(D8$$SyntheticClass:0)
W/default : 	at org.qtproject.qt.android.QtThread$1.run(QtThread.java:25)
W/default : 	at java.lang.Thread.run(Thread.java:1012)
W/default : Caused by: android.os.RemoteException: 
W/Qt A11Y : AccessibilityEvent with empty description
W/Qt A11Y : AccessibilityEvent with empty description

HTRamsey, 2025-02-19:
Permissions issue

xzxzhao2013, 2025-02-20:
But I have no problem with QGC4.4.3 on the same pad

ginolee218, 2025-02-20:
I can try use target sdk = 28

xzxzhao2013, 2025-02-21:
Thank you, I changed it to 28 and the serial port can be used

closed 2025-02-21 by HTRamsey