mavlink/qgroundcontrol #12492: latest master "qgc.videomanager.videoreceiver.gstreamer.api: <h264parse10> broken/invalid nal Type: 1 Slice,"

lida2003, 2025-02-22:
## Expected Behavior
Get H264 UDP stream output

## Current Behavior
No video

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. build qgc +qt6.8.2 on ubuntu 22.04
2.`gst-launch-1.0 udpsrc port=5600 ! application/x-rtp ! rtpjitterbuffer ! rtph264depay ! avdec_h264 ! videoconvert ! autovideosink` works for h264 video stream
3. qgc use udp h264 failed

## System Information
When posting bug reports, include the following information
- Operating System: NVIDIA linux L4T 36.4
- QGC Version: master
```
commit d185cea71ad1f36f20f909c4620658bb995cab3a (HEAD -> master, origin/master, origin/HEAD)
Author: Hamish Willee <hamishwillee@gmail.com>
Date:   Thu Feb 20 16:50:25 2025 +1100

    Move docs crowdin file to root (#12484)

    * Rename crowdin yet again

    * Attempt to fix base path

    * Try root level
```
- QGC build: self-built from source
- Flight Controller: NA
- Autopilot (with version): NA

HTRamsey, 2025-02-22:
EGL is untest/unsupported at the moment

lida2003, 2025-02-22:
> EGL is untest/unsupported at the moment

How can i disabled the egl, use gst by default? Or where can i find the code where needed to be changed and tested?

HTRamsey, 2025-04-30:
make sure you do: sudo apt install nvidia-l4t-jetson-multimedia-api

closed 2025-07-04 by HTRamsey