mavlink/qgroundcontrol #12506: [Android] UI Adaptation Failure and Scroll Malfunction on Small-screen Devices

OutSider070, 2025-02-25:
## Expected Behavior
1. QGC interface should properly adapt to different screen sizes/resolutions.
2. All interface elements should be accessible via vertical/horizontal scrolling in settings pages.

## Current Behavior
1. UI Adaptation Issues on small-screen Android devices (RC controllers, phones):
Oversized icons
Truncated interface elements

2. Scroll Functionality Failure in application settings pages:
Vertical scrolling completely non-functional
Horizontal scrolling unavailable when needed
(Note: Map view scrolling works normally)

3. Consistent reproduction across different device types:
Physical RC controllers
Mobile phones
Android emulators

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Build Android APK from master branch using:
Qt 6.6.3
JDK-17
NDK 26.3.11579264
SDK Platform 35
2. Install on small-screen Android device (e.g. phone or RC controller)
3. Navigate to any settings page requiring scrolling
4. Attempt vertical/horizontal scrolling

## System Information
- Operating System: Android 8.1,  Android 15
- QGC Version: Latest master branch
- QGC build: self-built from source
- Flight Controller: N/A
- Autopilot (with version): N/A

## Detailed Description
The combination of UI adaptation failures and scroll malfunction severely impacts usability on compact Android devices. Key observations:
Settings pages become partially inaccessible due to failed scrolling
Issue persists across different input methods (touchscreen, emulator mouse)

## Log Files and Screenshots

[image]

[image]

HTRamsey, 2025-02-25:
Funny, I just found that too. A possible fix is here: https://github.com/mavlink/qgroundcontrol/blob/112bbd9d7b99c30bc0ac4c8aa50a55ed51a46e5a/src/QmlControls/QGCFlickable.qml#L12

OutSider070, 2025-02-25:
> Funny, I just found that too. A possible fix is here:
> 
> [qgroundcontrol/src/QmlControls/QGCFlickable.qml](https://github.com/mavlink/qgroundcontrol/blob/112bbd9d7b99c30bc0ac4c8aa50a55ed51a46e5a/src/QmlControls/QGCFlickable.qml#L12)
> 
> Line 12 in [112bbd9](/mavlink/qgroundcontrol/commit/112bbd9d7b99c30bc0ac4c8aa50a55ed51a46e5a)
> 
>  maximumFlickVelocity:   (ScreenTools.realPixelDensity * 25.4) * 8   // About two inches per second

@HTRamsey I tried commenting out this line of code, but it didn't make any difference.

HTRamsey, 2025-02-25:
Huh interesting because that fixed it completely for me. We'll see what Don comes up with

HTRamsey, 2025-02-27:
Should be resolved by #12517. Reopen if not

closed 2025-02-27 by HTRamsey