mavlink/qgroundcontrol #12509: Ubuntu 24.04 gstreamer video receiver (via dynamic rtsp stream uri) fails to open resource

dirksavage88, 2025-02-25:
## Expected Behavior
RTSP streaming (receiving rtsp URI) should work out of the box when adhering to the camera protocol.

## Current Behavior
Gstreamer receiver fails with:

`
VideoReceiverLog: GStreamer error: Could not open resource for reading and writing.
`

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. On an ubuntu 24.04 machine, with voxl2 running voxl-px4, attach to the udp instance of the flight controller
2. With RTSP streaming enabled in voxl streamer and configuration mirrored in voxl-mavcam manager, assign the correct IP address in the rtsp uri config
3. QGC in linux fails to show the rtsp stream

## System Information

- Operating System: Ubuntu 24.04, kernel 6.8.0-53-generic
- QGC Version: 4.4.3 stable, 4.4.2, Linux Daily 112bbd9 64 bit
- QGC build: Daily & stable
- Flight Controller: voxl2
- Autopilot (with version): voxl-px4

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots

[image]

[qgcbug.txt](https://github.com/user-attachments/files/18970489/qgcbug.txt)

HTRamsey, 2025-02-27:
I think this must be related to setting an address from a mavlink camera automatically since it works for me with a manually entered rtsp stream

dirksavage88, 2025-03-12:
@HTRamsey any luck on reproducing this? I can verify the manual input streams work, but when there is an offboard camera protocol that sends the rtsp uri to QGC (camera protocol v2) it breaks. Went back to QGC 4.4.3 and this feature worked fine

HTRamsey, 2025-03-13:
Should be simple enough to fix once I can get a mavlink camera setup running,

closed 2025-04-30 by HTRamsey