mavlink/qgroundcontrol #12526: 💖💖💖关于解决QGC4.4.3中文BUG，无法标注航点的问题💖💖💖

yuxinqing123, 2025-03-01:
About solving the problem that QGC4.4.3 Chinese BUG cannot mark waypoints
系统：ubuntu22.04
QGC：v4.4.3
QT：15.5.2

QGC以中文启动时无法标记航点，只有在英文状态下启动然后转换为中文才能标记航点。这个BUG源于中文翻译脚本里面以中文逗号作为多个选项的分隔符，导致编译器认为这个只是一个整体。所以翻译时数量不匹配，导致BUG。

解决方法：将所有分隔的中文逗号都修改为英文逗号。主要需要修改的文件有两个：qgc_json_zh_CN.ts和qgc_source_zh_CN.ts.可以在QT中进行搜索，然后在文件中查找中文逗号全部替换为英文逗号。虽然只有所谓分隔符的中文逗号才会造成影响，但是比较难找，所以干脆全部替换。

[image]

但是还存在一个问题，就是这个文件里面有一处中文翻译是以顿号作为分隔符，所以需要将这里顿号改为英文逗号。之后就解决了！！！

[image]

[image]

💕💕💕如果对你有帮助就顶我上去，让更多人看到，感谢！！！💖💖💖💖

DonLakeFlyer, 2025-03-01:
This leading space thing is fixed in daily. Let me find it and bring it across to 4.4

hushidong, 2025-03-11:
感谢，厉害！

PR #12530 (2025-03-03): Trim enumStrings/enumValues
* Translators have a tendency to add spacing after the commands which breaks things
* Fix #12526
closed 2025-04-22 by HTRamsey