mavlink/qgroundcontrol #12537: QGC daily broken on windows 11 (does not open, spams windows command terminals)

dirksavage88, 2025-03-04:
## Expected Behavior
QGC daily on windows 11 opens normally
## Current Behavior
QGC daily on windows 11 causes a terminal to open and close repeatedly, spamming and causing issues with other open applications

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Download QGC Daily for windows: https://docs.qgroundcontrol.com/master/en/qgc-user-guide/releases/daily_builds.html
2. In windows 11, run the install application
3. Open QGC daily
4. Observer terminal open/close spamming

## System Information
When posting bug reports, include the following information
- Operating System: Windows 11
- QGC Version: daily
- QGC build: daily
- Flight Controller: none
- Autopilot (with version): none

DonLakeFlyer, 2025-03-04:
I think this is gst_plugin_scanner.exe running over and over again and failing every time.

HTRamsey, 2025-03-04:
Yup I noticed that too but I don't know what causes it right now. Also doesn't happen on all my systems

dirksavage88, 2025-03-05:
@HTRamsey is there any way I can troubleshoot further? I only recently (last 2 weeks) starting seeing this issue in the daily release for windows 11.

HTRamsey, 2025-03-05:
I think it may be from switching the GStreamer version to 1.24 instead of 1.22. I don't recall any other recent changes to the GStreamer stuff but I will double check today. 

I will make a PR to switch it back and we can test that

AlexKlimaj, 2025-03-12:
Seeing the same thing here.

HTRamsey, 2025-03-13:
Can one of you guys try the attached PR that uses a different version of GStreamer? @DonLakeFlyer is correct it is gst_plugin_scanner.exe related

PR #12539 (2025-03-13): CI: Change Windows GStreamer back to 1.22
Attempt to fix #12537
closed 2025-03-13 by HTRamsey