mavlink/qgroundcontrol #12546: Mission transfer failed with "Hover and Capture" option set in survey mission.

jacobguenther, 2025-03-09:
## Expected Behavior
The mission should transfer to the copter when the Hover and Capture option is set.

## Current Behavior
A pop up appears
```
Mission transfer failed. Error: Param 4 invalid
value. Item #4 Command: Start image capture
Value: nan
```

## System Information
- Operating System: Pop!_OS 22.04
- QGC Version:  4.3.0
- QGC build: Flatpak
- Flight Controller: Pixhawk 6c
- Autopilot (with version): Ardupilot Copter 4.5.7

## Detailed Description
I checked the .plan file and saw
```
{
    "autoContinue": true,
    "command": 2000,
    "doJumpId": 4,
    "frame": 2,
    "params": [
        0,
        0,
        1,
        null,
        null,
        null,
        null
    ],
    "type": "SimpleItem"
},
```

Looking at the MavLink docs
https://mavlink.io/en/messages/common.html#MAV_CMD_IMAGE_START_CAPTURE
The fourth parameter to command 2000 Image start capture should be 0 or 1.

When I manually change the value to 0 or 1 from null the mission successfully uploads. For the survey mission it should probably be set to 1 though.

DonLakeFlyer, 2025-03-09:
I'm pretty sure this is fixed in daily builds. Can you try that?

jacobguenther, 2025-03-09:
Checked on the 4.4.3 appimage and it works fine. Thanks. Didn't see anything about it in the change logs so I assumed it would still be broken.

DonLakeFlyer, 2025-03-09:
Can you verify a daily build as well?

DonLakeFlyer, 2025-03-10:
@HTRamsey Can you look at the appimage thing?

DonLakeFlyer, 2025-03-11:
Re-open for the AppImage problem

HTRamsey, 2025-03-13:
GCC version needs to be changed to 11 https://doc.qt.io/qt-6/linux.html

jacobguenther, 2025-03-14:
I upgraded my OS on my laptop to 24.04 and the daily AppImage works on that. But I still have the same problem with the parameter being null when uploading a survey mission with hover and capture.

DonLakeFlyer, 2025-03-15:
Can you attach a plan file that is failing? Change extension to .txt to let github attach it.

jacobguenther, 2025-03-16:
[coast_guard_beach_muskeg.txt](https://github.com/user-attachments/files/19275385/coast_guard_beach_muskeg.txt)

A survey mission with a polygon survey area. And the "Options" -> "Hover and Capture" checked. When uploading a dialog pops up with "Mission transfer failed. Error: Param 4 invalid value. Item #5 Command: Start image capture Value: nan"

Any mission with Hover and Capture seams to fail for me.

DonLakeFlyer, 2025-03-17:
Thanks for the bug report and help

DonLakeFlyer, 2025-03-18:
@jacobguenther Can you validate this is fixed for you now? You will need to create a new plan

jacobguenther, 2025-03-21:
Created a new plan with "Hover and Capture". It uploads without error. Thanks for handling this. It probably would have taken me hours to find where it was set in the code.
Now to test the actual behavior is right.

atifhalim, 2025-05-10:
i am using the stable QGroundcontrol version v4.4.4.  Created a new plan with "Hover and Capture" under the Survey waypoint.  receive the following error:  Pram 4 invalid value.  item #6 command: start image capture value: nan

Ask an LLM and it think the issue is the following:  
The error message you received from QGroundControl (QGC), "Param 4 invalid value. item #6 command: start image capture value: nan," indicates that the mission upload fails because the CAMERA_TRIGGER command (MAVLink command ID 2000) at mission item #6 contains an invalid value (NaN) in Param4. This is a known issue in some QGC versions where the CAMERA_TRIGGER command (used for "Hover and Capture" in surveys) incorrectly sets Param4 to NaN or null, which the ArduCopter firmware rejects. The .plan file you provided confirms this: multiple CAMERA_TRIGGER commands (e.g., item #6, doJumpId: 6) have params: [0, 0, 1, null, null, null, null], where Param4 is null, causing the invalid value error.

HTRamsey, 2025-05-10:
@atifhalim this is fixed in the master build

atifhalim, 2025-05-11:
> [@atifhalim](https://github.com/atifhalim) this is fixed in the master build

@HTRamsey thanks for the replay.  However, the master branch has other issues.  For instance, RTSP video stream does not really work.  Perhaps I should ask the LLM to fix the master branch  :)

HTRamsey, 2025-05-11:
If you experience a problem with the master branch then please create an issue with details like your platform and whether you are building it yourself or downloading the release. As far as RTSP, I do not personally have any issues on Linux/Android/Windows nor do any of the customers I deploy to, so more details would be helpful

PR #12571 (2025-03-17): Survey: Fix hover and capture camera triggering
Fix #12546