mavlink/qgroundcontrol #12547: Connecting Bluetooth Xbox controller on PX4 SITL Ubuntu 24.04 causes QGC to not respond

sickopickle, 2025-03-09:
## Expected Behavior
When I connect the controller it should function as normal and not cause QGC to freeze.

## Current Behavior

When I connect the controller it causes QGC to freeze and not respond.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Install Gazebo Harmonic/ROS Jazzy on Ubuntu 24.04
2. Install PX4 on Ubuntu 24.04
3. Install QGroundControl
4. Run the example `make px4_sitl gz_x500`

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 24.04
- QGC Version: latest
- QGC build: Daily https://docs.px4.io/main/en/dev_setup/dev_env_linux_ubuntu.html 
- Flight Controller: N/A
- Autopilot (with version): PX4 1.15.4 (latest)

## Log Files and Screenshots
- QGC Console Logs - https://pastebin.com/GRgFjmjB
- Autopilot logs - https://pastebin.com/gACFM2hg

[image]

HTRamsey, 2025-03-10:
Bit hard to diagnose specific controllers, but we can always test SDL3 and see if that fixes it

HTRamsey, 2025-03-10:
I just did a calibration with a UXV controller about a week ago without issue, but I can check with a wireless steelseries later today

DonLakeFlyer, 2025-03-10:
Wait, my bad. I had it in the wrong mode. It's working fine.

sickopickle, 2025-03-12:
I bought a Logitech Extreme 3D Pro and it works fine out of the box. I guess my controller isn't supported (it's not really a normal Xbox controller, more of a cheap third-party Nintendo controller, although I think its recognized by the system as a Xbox controller).

closed 2025-03-12 by sickopickle