mavlink/qgroundcontrol #12548: QGC master daily build doesn't install on Android device

bnsgeyer, 2025-03-10:
## Expected Behavior
QGC daily build installs on Android device.  

## Current Behavior
Device reports application not installed.  This was tested on a Vantage Robotics Vision 2 GCS running Android version 10.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.   Tap on QGroundControl.apk file
2.   Confirm desire to install
3.   shows progress bar but then reports App did not Install
4.

## System Information
When posting bug reports, include the following information
- Operating System: Android 10
- QGC Version: Master
- QGC build:  daily
- Flight Controller: cube orange plus
- Autopilot (with version): ArduCopter 4.5.7

## Detailed Description
Pretty self explanatory

DonLakeFlyer, 2025-03-11:
Hmm, working for me on my Samsung. @HTRamsey Was there another report on android 10 installs failing? I vaguely remember something.

bnsgeyer, 2025-03-11:
@DonLakeFlyer i’m trying to see if this version of software will solve my issue with accessing the gimbal joystick on this GCS controller. Would it matter whether I get a 64 bit version or not? There was only one selection under the daily builds for the android devices. Thanks!

DonLakeFlyer, 2025-03-12:
@HTRamsey Can't remember is that APK published by CI multi-abi, 32 and 64?

bnsgeyer, 2025-03-12:
@DonLakeFlyer any chance you could make a 64 bit version available for me to download.

HTRamsey, 2025-03-13:
It is multi-abi, yeah. Also, I do recall something else about Android 10 maybe? I think it has something to do with the set compile sdk

DonLakeFlyer, 2025-03-15:
I don't have an Android 10 device to debug on. Need to stare at logcat while the APK is being installed to get error messages out of it.

bnsgeyer, 2025-03-15:
@DonLakeFlyer if you can make a 64 bit android target, I would be willing to test.

DonLakeFlyer, 2025-03-15:
The APK posted for daily is multi-abi so it includes a 64 bit version. Have you ever used Android SDK tools to debug an app before?

bnsgeyer, 2025-03-15:
> Have you ever used Android SDK tools to debug an app before?

@DonLakeFlyer no. I have not.

bnsgeyer, 2025-03-19:
@DonLakeFlyer What specific tool would you like me to use to debug the app during install?  You may have to hold my hand a little to get me started.  From what I can tell, I need to download and install android studio on my windows machine and then put the android device in developer mode?  This allows me to connect the android device to the windows machine to perform debugging while the android device installs the app?   How to I log the messages while it performs the install?

DonLakeFlyer, 2025-03-19:
Basically that. If you are willing I'll put together some instructions tomorrow morning. Thanks for the help.

bnsgeyer, 2025-03-19:
@DonLakeFlyer yes I would be willing to perform the test. Please send me the instructions. Thanks!

DonLakeFlyer, 2025-03-19:
So you can follow these instructions to install android adb and set up your device for debugging: https://www.howtogeek.com/125769/how-to-install-and-use-abd-the-android-debug-bridge-utility/.

Once you have that all setup and you can see your device with `adb devices`:
* Click the apk open the install page, don't click install yet
* In the terminal run 'adb logcat >log.txt`
* Now click install
* When it fails Ctrl+C the running `adb`
* Attach the log here

bnsgeyer, 2025-03-21:
@DonLakeFlyer Here is the log from the attempted installation of today's daily build.  If the is anything else you need me to do, let me know.

[log.txt](https://github.com/user-attachments/files/19389154/log.txt)

DonLakeFlyer, 2025-03-21:
```
03-21 17:35:15.934  1342  1502 W PackageManager: Package org.mavlink.qgroundcontrol signatures do not match previously installed version; ignoring!
```

You are going to have to delete your previous version of QGC from the device and then install again from apk.

bnsgeyer, 2025-03-21:
@DonLakeFlyer not sure what you mean. Do I need to delete the custom version or the open source version with was 4.4.3?

bnsgeyer, 2025-03-25:
@DonLakeFlyer I uninstalled the custom QGC version installed by Vantage Robotics and then I was able to successfully install the Open source daily build of QGC.  Thanks for you help!

closed 2025-03-25 by bnsgeyer