mavlink/qgroundcontrol #12549: Custom modes not shown

dakejahl, 2025-03-11:
PX4 allows users to define "custom modes" from a companion computer according to the specification here:
https://docs.px4.io/main/en/ros2/px4_ros2_control_interface.html

This worked up until maybe 6 months ago, at which some point something changed in QGC and this no longer works. I spent maybe an hour looking into this and this is what I discovered.

**Output from QGC console**
```
StandardModesLog: Completed, num modes: 25
StandardModesLog: Available modes changed, re-requesting
StandardModesLog: Requesting available modes
StandardModesLog: Got mode: "Manual" , idx: 1 , custom_mode 65536
StandardModesLog: Got mode: "Altitude" , idx: 2 , custom_mode 131072
StandardModesLog: Got mode: "Position" , idx: 3 , custom_mode 196608
StandardModesLog: Got mode: "Mission" , idx: 4 , custom_mode 67371008
StandardModesLog: Got mode: "Hold" , idx: 5 , custom_mode 50593792
StandardModesLog: Got mode: "Safe Recovery" , idx: 6 , custom_mode 84148224
StandardModesLog: Got mode: "Position Slow" , idx: 7 , custom_mode 33751040
StandardModesLog: Got mode: "Acro" , idx: 8 , custom_mode 327680
StandardModesLog: Got mode: "Termination" , idx: 9 , custom_mode 655360
StandardModesLog: Got mode: "Offboard" , idx: 10 , custom_mode 393216
StandardModesLog: Got mode: "Stabilized" , idx: 11 , custom_mode 458752
StandardModesLog: Got mode: "Takeoff" , idx: 12 , custom_mode 33816576
StandardModesLog: Got mode: "Land" , idx: 13 , custom_mode 100925440
StandardModesLog: Got mode: "Follow Target" , idx: 14 , custom_mode 134479872
StandardModesLog: Got mode: "Precision Landing" , idx: 15 , custom_mode 151257088
StandardModesLog: Got mode: "Orbit" , idx: 16 , custom_mode 16973824
StandardModesLog: Got mode: "VTOL Takeoff" , idx: 17 , custom_mode 168034304
StandardModesLog: Got mode: "CustomMode" , idx: 18 , custom_mode 184811520
StandardModesLog: Got mode: "(Mode not available)" , idx: 19 , custom_mode 201588736
StandardModesLog: Got mode: "(Mode not available)" , idx: 20 , custom_mode 218365952
StandardModesLog: Got mode: "(Mode not available)" , idx: 21 , custom_mode 235143168
StandardModesLog: Got mode: "(Mode not available)" , idx: 22 , custom_mode 251920384
StandardModesLog: Got mode: "(Mode not available)" , idx: 23 , custom_mode 268697600
StandardModesLog: Got mode: "(Mode not available)" , idx: 24 , custom_mode 285474816
StandardModesLog: Got mode: "(Mode not available)" , idx: 25 , custom_mode 302252032
StandardModesLog: Completed, num modes: 25
qml: hiddenFlightModesList:
qml: [Manual,Offboard,Rattitude,Precision Land]
```

Notice the "CustomMode" in that list. This is from my application I am testing against. I also added a console.log to show the hidden modes, just to check if we're somehow filtering out our new CustomMode.

Here's what QGC shows me:
[image]

So the modes QGC shows don't match what's actually available. There isn't even a "Return" mode anymore since it was renamed to "Safe Recovery".

HTRamsey, 2025-03-17:
@dakejahl does this have anything to do with how you can now edit shown modes after expanding that window?

dakejahl, 2025-03-17:
> [@dakejahl](https://github.com/dakejahl) does this have anything to do with how you can now edit shown modes after expanding that window?

Nope totally unrelated. I think the mode list is hardcoded somewhere instead of using the mode list that is provided by the flight controller

DonLakeFlyer, 2025-03-17:
The mode list should be dynamically being updated using the new flight mode mavlink protocol code. It's just not working correctly. It was likely broken by the work to not reference standard flight modes by string names.

DonLakeFlyer, 2025-03-17:
So how did you get this step for ROS2 install:
```
pip install --user -U empy==3.3.4 pyros-genmsg setuptools
```

To work on Ubuntu 24?

dakejahl, 2025-03-17:
I'm still on Ubuntu 22.04 but it should be as easy as 
```
sudo apt-get install ros-humble-desktop-full
```

DonLakeFlyer, 2025-03-18:
Also FYI: "Safe Recovery" is just the strange internal name for the standard mode MAV_STANDARD_MODE_SAFE_RECOVERY. This is the "Return" flight mode in PX4. No idea what it has such a strange name in the new protocol given the prevalence of RTL/Return terminolofy. The protocol isn't meant to be used to rename known standard modes. Hence, it's still "Return" in QGC and also translated to other languages as needed.

DonLakeFlyer, 2025-03-18:
I"ll take another stab at this tomorrow

dakejahl, 2025-03-18:
Yeah, I'm not a fan of the name either
https://github.com/mavlink/mavlink/pull/2191

closed 2025-03-25 by DonLakeFlyer