mavlink/qgroundcontrol #12560: QGC Daily Gimbal Interface Does not open (but gimbal test commands work) (Windows and Linux daily)

dirksavage88, 2025-03-13:
## Expected Behavior
Gimbal device attitude status is being streamed, we are sending a gimbal heartbeat, and the gimbal manager on the autopilot is enabled. MNT_MODE_IN and MNT_MODE_OUT are set to gimbal protocol v2. We should see the gimbal interface populate in QGC

## Current Behavior
Windows daily and Linux daily I do not see the gimbal interface populate. I see the compid of the gimbal and I can successfully send gimbal test commands in PX4 (they work). 

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Download QGC linux or windows daily
2. Enable the gimbal manager in PX4
3. Set MNT_MODE_IN and MNT_MODE_OUT to gimbal protocol v2
4. Observe if gimbal interface populates (assuming a gimbal that speaks mavlink and can send device attitude status and receive set device attitude commands)

## System Information
When posting bug reports, include the following information
- Operating System: Windows 11 & Ubuntu 24.04
- QGC Version: Daily
- QGC build: Daily
- Flight Controller: Voxl2/Voxl2 mini
- Autopilot (with version): PX4 1.14

## Detailed Description

## Log Files and Screenshots

DonLakeFlyer, 2025-03-14:
Save a telemetry log which shows the problem. Put it up on the internet somewhere and point me at it so I can debug.

DonLakeFlyer, 2025-04-16:
@dirksavage88 Telemetry log?

DonLakeFlyer, 2025-05-05:
@Davidsastresas Can you repro this?

jnomikos, 2025-05-07:
I built from master and with a gremsy vio and ardupilot system, I am able to see the gimbal controls show up. However, I notice a few issues on the vio.

1. There is like a 2 second delay between when I press a control button (tilt 90 or center) and when the gimbal responds. This does not happen if I modify the code to send GIMBAL_DEVICE_SET_ATTITUDE instead:
```
void GimbalController::sendPitchBodyYaw(float pitch, float yaw, bool showError)
{
    if (!_tryGetGimbalControl()) {
        return;
    }

    _rateSenderTimer.stop();
    _activeGimbal->setAbsolutePitch(0.0f);
    _activeGimbal->setYawRate(0.0f);

    // qCDebug(GimbalControllerLog) << "sendPitch: " << pitch << " BodyYaw: " << yaw;

    const unsigned flags = GIMBAL_MANAGER_FLAGS_ROLL_LOCK
                         | GIMBAL_MANAGER_FLAGS_PITCH_LOCK
                         | GIMBAL_MANAGER_FLAGS_YAW_IN_VEHICLE_FRAME;

    WeakLinkInterfacePtr weakLink = _vehicle->vehicleLinkManager()->primaryLink();
    if (!weakLink.expired()) {
        mavlink_message_t       msg;
        SharedLinkInterfacePtr  sharedLink = weakLink.lock();

        float q[4];
        mavlink_euler_to_quaternion(qDegreesToRadians(0.0f), qDegreesToRadians(pitch), qDegreesToRadians(yaw), q);

        mavlink_msg_gimbal_device_set_attitude_pack_chan(
            1,
            MAV_COMP_ID_ONBOARD_COMPUTER,
            sharedLink->mavlinkChannel(),      // channel
            &msg,
            1,
            154,
            flags, q, NAN, NAN, NAN
        );

        _vehicle->sendMessageOnLinkThreadSafe(sharedLink.get(), msg);
    }
}
```

2. When using Tilt 90, the gimbal yaws 180 degrees. This appears to be an issue on gremsy gimbals in general. They can't tilt 90 degrees, more like 89.5. If I make the tilt 90 button tilt 89 degrees, this issue goes away

3. "Point Home" makes the gremsy vio go unresponsive and I can't use any other button to control the vio after clicking it until rebooting drone. Yaw follow and yaw lock still work though after clicking "point home"

Should I create a new issue for these?

jnomikos, 2025-05-07:
@dirksavage88 

Was your gimbal sending GIMBAL_MANAGER_INFORMATION or GIMBAL_MANAGER_STATUS after the GCS requested it?

dirksavage88, 2025-05-07:
@jnomikos I am using the px4 gimbal manager and having autopilot take the role of the manager, and only sending gimbal device attitude status. But I am using a compid of 154 for the gimbal device. 

The gimbal device is also sending heartbeat and responding to requests for gimbal device info (MAV_CMD_REQUEST_MESSAGE)

Davidsastresas, 2025-05-10:
I just tested this in latest QGC master, and Ardupilot master, and it seems to work fine.

I think the issues reported by @dirksavage88 are probably PX4 related, so I would double check with anybody in the PX4 side.

The issues reported by @jnomikos seem to be independent of the Gremsy driver for Ardupilot. I would say it is probably independent to VIO camera handling. I've seen in the past Gremsy not re-using the implementations already working for them. For example we recently found an issue with their new Sony ILX gimbals, and it was fixed a few weeks back. I would not be surprised if this is the case too with VIO.

I have a VIO gimbal around, and I will work with it on the next few weeks, so I will have the chance to double check that.

In any case, I don't think none of the issues mentioned here are a problem specific to QGC, so I would suggest to close this issue.

jnomikos, 2025-05-14:
> [@DonLakeFlyer](https://github.com/DonLakeFlyer) I just tested this in latest QGC master, and Ardupilot master, and it seems to work fine.
> 
> I think the issues reported by [@dirksavage88](https://github.com/dirksavage88) are probably PX4 related, so I would double check with somebody in the PX4 side.
> 
> The issues reported by [@jnomikos](https://github.com/jnomikos) seem to be independent of the Gremsy driver for Ardupilot. I would say it is probably independent to VIO camera handling. I've seen in the past Gremsy not re-using the implementations already working for them. For example we recently found an issue with their new Sony ILX gimbals, and it was fixed a few weeks back. I would not be surprised if this is the case too with VIO.
> 
> I have a VIO gimbal around, and I will work with it on the next few weeks, so I will have the chance to double check that.
> 
> In any case, I don't think none of the issues mentioned here are a problem specific to QGC, so I would suggest to close this issue.

I agree with this and have found similar stuff with my testing. And (in my opinion) the best we can do with gremsy is just report to them what isn't standard.

closed 2025-05-14 by DonLakeFlyer