mavlink/qgroundcontrol #12564: FOLLOW_TARGET message sent at extreme rate

tpwrules, 2025-03-14:
## Expected Behavior
When enabling the "Stream GCS Position" function using a NMEA GPS in "Always" mode, QGC should send the FOLLOW_TARGET message perhaps a few times per second, or at the rate of the data coming from the GPS.

## Current Behavior

When enabling the "Stream GCS Position" function using a NMEA GPS in "Always" mode, QGC tries to send the FOLLOW_TARGET message tens of thousands of times per second. This saturates the vehicle comm link and eventually fills up the hard disk by creating MAVLink telemetry logs at a dozen megabytes a second.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Set up NMEA GPS device in the AutoConnect settings
2. Set Stream GCS Position mode to Always
3. Connect to vehicle
4. Connect GPS to computer
5. Observe connection becomes sluggish and telemetry log grows at an extreme rate.

## System Information
When posting bug reports, include the following information
- Operating System: Xubuntu 22.04
- QGC Version: 4.4.3
- QGC build: stable
- Flight Controller: Cube Orange Plus
- Autopilot (with version): ArduCopter 4.5.4

## Detailed Description
The NMEA GPS is a Holybro UM982 module and is configured to send data at 1Hz.

I think the root cause of the bug is this function:

https://github.com/mavlink/qgroundcontrol/blob/0243afc521428855b842df9e4fff1b7af8ed4b7e/src/FollowMe/FollowMe.cc#L79-L85

The `updateInterval()` defaults to 0, so the `qMin` will set the timer for 0 milliseconds, evidently causing the function to send the data to be called as fast as possible.

I think the `qMin` should instead be `qMax`. A 250 millisecond (value of `kMotionUpdateInterval`) lower bound is quite reasonable. I am happy to send a PR to fix this if my assessment is accurate.

## Log Files and Screenshots

From the ArduPilot stream stats web tool:

[image]

DonLakeFlyer, 2025-03-14:
Can you debug on your system why this: https://github.com/mavlink/qgroundcontrol/blob/master/src/PositionManager/PositionManager.cpp#L215 is not updating the interval to be above 0?

I don't disagree with your assessment of an upper version lower bound. Just want to make sure it's understood fully as to why it's so out of whack as it currently stands.

tpwrules, 2025-03-14:
I suspect, without having debugged but only studying the code, that there must be some race condition specifically in "Always" mode where that variable is not initialized yet.

Searching a bit more, the value that variable is set to seems pretty arbitrary: https://github.com/qt/qtpositioning/blob/17886fb464ec1aff61cd8631dd5ef7c400602379/src/positioning/qnmeasatelliteinfosource.cpp#L482 . I am using the AppImage so whatever version of Qt is included in there, but that value has been consistent for a long time. I don't see a case where that line would set the variable to a value other than 2, so that leads me to believe the race condition theory.

I can see the 500 messages per second when using the "only in follow me" option, but that is still a quite excessive amount. Just less likely to completely sink the ship, so maybe it has gone unnoticed. I also suspect most users use a UBlox GPS.

DonLakeFlyer, 2025-03-14:
Yeah, its a race condition in the always code path. The coupling of the FollowMe code and the position manager stuff is just a mess. That really needs to be cleaned up.

PR #12781 (2025-05-08): Follow Me: Don't trust update interval from device
* Fix #11245
* Fix #12564
closed 2025-05-08 by DonLakeFlyer